twrpdtgen/twrpdtgen #3: Not packing properly

mrmazakblu, 2020-08-10:
I know it says Android 10 is beta.
I know recovery is not fully working on Android 10.

When I build twrp. The image is not packed correctly. I manually unpack it, add the correct flags and files to image, and it will boot.

Stock has both ::
recovery.img-dtb
recovery.img-recoverydtbo

Built recovery only has ::
recovery.img-recoverydtbo

There are some differences in recovery.img-dtboff and recovery.img-secondoff

Device == Blu g90 g0310ww

Device dump with recovery img

https://github.com/mrmazakblu/blu_g0310ww_dump

Device tree 
https://github.com/mrmazakblu/Blu_G90_G0310WW-twrp

I think I'm making some noob mistake, maybe you can help

SebaUbuntu, 2020-08-10:
It seems the script isn't able to check if a dtb exists
I will fix it

SebaUbuntu, 2020-08-10:
Fixed, please retry

SebaUbuntu, 2020-08-10:
Force pushed, I forgot a line

mrmazakblu, 2020-08-10:
Umm, that seems to have picked up the Dt.img, but made other issues. 
Now the offset calculations are ;; 
0x0x
instead of ;;
0x

then manually removed the extra "0x" and then build gives error 
"header_version and dt cannot be set at same time"

mrmazakblu, 2020-08-10:
I have not verified , but I will soon. 
I think AIK is not properly unpacking the recovery.img.
on windows I use carliv kitchen and it does not unpack correctly either. But when I use ART it does unpack /repack correct. 

ART -android ROM tool.

SebaUbuntu, 2020-08-11:
- Fixed 0x issue
- Fixed DTBO inclusion

Sorry for that, I had to adapt to AIK 3.7 changes

SebaUbuntu, 2020-08-11:
> "header_version and dt cannot be set at same time"

Could you send the entire error with del.dog

mrmazakblu, 2020-08-11:
> > "header_version and dt cannot be set at same time"
> 
> Could you send the entire error with del.dog?

this is the error.

https://del.dog/ackugheloc

mrmazakblu, 2020-08-11:
I realize there are some things that are different for building in Android 10 source. Like in ten source there needs to be /system/etc folder in device tree. 

There is no longer vendorsetup.sh either.

Like I said this is big can of worms opened up here.

SebaUbuntu, 2020-09-30:
Fixed them all in new Python rewrite

closed 2020-09-30 by SebaUbuntu

mrmazakblu, 2020-09-30:
Just used fresh got pull, and built new tree for current device. Ulefone note 9p (almost identical to Umidigi a9 pro and Blu G90).

The script still gives "TARGET_2ND_ARCH_VARIANT. := arm7-a-neon"
Building on 10 source throws an error with this.do not recall exact message.

2 more things I think are , maybe not working

"dtb_offset" did not get written
.
 I was expecting to see 
Board_mkbootimg_args += --dtb $(target_prebuilt_dtb)

But what was written by program was

Board_mkbootimg_args += --dt $(target_prebuilt_dtb)

SebaUbuntu, 2020-10-01:
About dtb_offset, it's not necessary, but it may help

https://github.com/SebaUbuntu/TWRP-device-tree-generator/blob/master/start.py#L298-L299
It's impossible that it wrote --dt $(target_prebuilt_dtb), try to reclone instead of pulling

mrmazakblu, 2020-10-01:
>>>> must be working on things..... the file start.py is gone from repo. right now.

SebaUbuntu, 2020-10-02:
Recheck wiki, command has been updated

mrmazakblu, 2020-10-02:
Alright. 

I see on page 2 of wiki, new cmds. But on page 1 it still says 

" If it isn't already, make start.py executable"

mrmazakblu, 2020-10-02:
do i need to have any modules first. I am getting errors.

I tries=d in 2 differnt versions of python and get 2 different errors

https://del.dog/prehulosam

SebaUbuntu, 2020-10-02:
Wiki updated