twrpdtgen/twrpdtgen #9: Unable to get past extracting recovery

mrmazakblu, 2020-10-03:
Sorry, I have more errors with new .py tool.

when feeding an android 10 recovery.img i get an error "no expected binary found"

and when i tried a pie device recovery.img  i got "SameFileError" error

https://del.dog/uvexipofih

SebaUbuntu, 2020-10-04:
Send both recovery images

yshalsager, 2020-10-04:
@mrmazakblu for pie recovery, this error was fixed with 3478b229. Make sure you're using the latest tool before trying again.

mrmazakblu, 2020-10-04:
pie recovery was from BLU Bold-N1

https://github.com/mrmazakblu/blu_g0310ww_dump/blob/full_k62v1_64_bsp-user-10-QP1A.190711.020-1585383273-release-keys/boot.img

Q is from Ulefone Note 9P
https://github.com/mrmazakblu/ulefone_note_9p_dump/blob/full_k62v1_64_bsp-user-10-QP1A.190711.020-mp1V977-release-keys/boot.img

mrmazakblu, 2020-10-04:
sorry, Im doing too many things, those links are the boot.img

Recovery for pie device is 
https://github.com/mrmazakblu/blu_n0030ww_dump/blob/full_k71v1_64_bsp-user-9-PPR1.180610.011-1563880452-release-keys/recovery.img

the Q device recovery was too large to upload on github
https://www.androidfilehost.com/?fid=8889791610682951751

yshalsager, 2020-10-05:
> sorry, Im doing too many things, those links are the boot.img
> 
> Recovery for pie device is
> https://github.com/mrmazakblu/blu_n0030ww_dump/blob/full_k71v1_64_bsp-user-9-PPR1.180610.011-1563880452-release-keys/recovery.img
> 
> the Q device recovery was too large to upload on github
> d

Pie one works fine: https://del.dog/licyfusese.txt

And Q is fixed with e938be5 https://del.dog/ahytolerig.txt

closed 2020-10-05 by yshalsager