twrpdtgen/twrpdtgen #11: error: Can not locate config makefile for product "omni_Phoenix".

guosasipo, 2020-10-05:
log: https://del.dog/phinginone

recovery image file: https://drive.google.com/file/d/1p3FQ160e1INvgyNy-pyAg3uBi80hLQjP/view?usp=sharing

I dumped recovery image from my phone with stock firmware.
I have right repo.
Why do I get this error?

SebaUbuntu, 2020-10-05:
- Don't build as root
- Check that device tree is in device/samsung/Phoenix (check the uppercase P)

guosasipo, 2020-10-05:
> - Don't build as root
> - Check that device tree is in device/samsung/Phoenix (check the uppercase P)

That recovery image is based at android 10 can it be problem?

SebaUbuntu, 2020-10-05:
> That recovery image is based at android 10 can it be problem?

No, that shouldn't be a problem

guosasipo, 2020-10-05:
> > That recovery image is based at android 10 can it be problem?
> 
> No, that shouldn't be a problem

I solved tree generator didn't wrote androidproducts.mk

It have to write 

PRODUCT_MAKEFILES := \
$(LOCAL_DIR)/omni_Phoenix.mk

but tree generator wrote it

PRODUCT_MAKEFILES := \
$(LOCAL_DIR)/omni_.mk

I think you have to solve it

SebaUbuntu, 2020-10-05:
Don't close it if it isn't fixed

guosasipo, 2020-10-05:
> Don't close it if it isn't fixed

ok

SebaUbuntu, 2020-10-05:
https://github.com/SebaUbuntu/TWRP-device-tree-generator/commit/bc38bf675de1c48ca0421876c89585f63ea4df5b
Fixed

closed 2020-10-05 by SebaUbuntu

guosasipo, 2020-10-05:
> https://github.com/SebaUbuntu/TWRP-device-tree-generator/commit/bc38bf675de1c48ca0421876c89585f63ea4df5b
> Fixed

there is one more error

default recovery.fstab location set in twrp_dt_gen.py is

ramdisk/etc/recovery.fstab

but some recovery image's fstab file located in 

ramdisk/system/etc/recovery.fstab

so I had to edit it myself. I think you have to fix it too.