twrpdtgen/twrpdtgen #12: twrp_dt_gen.py issue

guosasipo, 2020-10-06:
there is one more error

default recovery.fstab location set in twrp_dt_gen.py is

ramdisk/etc/recovery.fstab

but some recovery image's fstab file located in

ramdisk/system/etc/recovery.fstab

so I had to edit it myself. I think you have to fix it too.

SebaUbuntu, 2020-10-06:
https://github.com/SebaUbuntu/TWRP-device-tree-generator/commit/ac544e5e2fd02e31ac0fd7fba41c189f57e47bbc
Fixed

And, https://github.com/SebaUbuntu/TWRP-device-tree-generator/commit/f8758a0e11726a54e27b5cdc70729dc172978ccb
Fixed system as root entry

closed 2020-10-06 by SebaUbuntu

mrmazakblu, 2020-10-06:
This , I do not think is fixed. 

I had same error, I recloned when I saw the patch. The tree gets generated without error, but the fstab I left in the tree root dir, and not linked in the boardconfig.mk

So will not be included in final product.

In my first few builds on TWRP -10 branch I had errors about second arch being set (arm v7 neon). This I suspect needs to be changed.

SebaUbuntu, 2020-10-06:
There's no need to mention where the fstab is in device tree, TWRP will automatically pick it