twrpdtgen/twrpdtgen #13: error at mka recovery image

guosasipo, 2020-10-06:
log https://del.dog/eravifeste
recovery image https://drive.google.com/file/d/1p3FQ160e1INvgyNy-pyAg3uBi80hLQjP/view?usp=sharing
tree from tree generator https://drive.google.com/file/d/1ZzP3sRTQq1HU_HF66ZzR6QaN19imgrLC/view?usp=sharing

How to solve it?

SebaUbuntu, 2020-10-06:
On terminal:
```
touch /home/user/twrp/out/target/product/Phoenix/root/../system/etc/selinux/plat_hwservice_contexts
touch /home/user/twrp/out/target/product/Phoenix/root/../system/etc/selinux/vendor_hwservice_contexts
```
And retry building, it's a TWRP 10 bug

guosasipo, 2020-10-06:
I had to wrote this command on terminal too

`touch /home/user/twrp/out/target/product/Phoenix/root/../vendor/etc/selinux/vendor_hwservice_contexts`

guosasipo, 2020-10-06:
build was successful
thanks

closed 2020-10-06 by guosasipo

mrmazakblu, 2020-10-06:
I have built TWRP for 3 device so far on TWRP 10.0, did not need to that. Maybe is so related, the tree used in this example is set to generic,. Mine built with cortex a53.

Just a guess

guosasipo, 2020-10-06:
> I have built TWRP for 3 device so far on TWRP 10.0, did not need to that. Maybe is so related, the tree used in this example is set to generic,. Mine built with cortex a53.
> 
> Just a guess

case by case