twrpdtgen/twrpdtgen #14: Python errors?

portsample, 2020-10-09:
First, Thank you very much for putting this together. I am very grateful for your efforts and hard work. I am building for a Sonim XP8. This is the error that I am getting,

    dfghom-gjhollo1:/home/glenn/TWRP-device-tree-generator # python3 -m twrpdtgen XP8A_ATT-   user-8A.0.5-11-8.1.0-10.54.00.zip
    TWRP device tree generator
    Python Edition
    Version 1.0.0

    AIK is up-to-date
    Getting device infos...
    Traceback (most recent call last):
      File "/usr/lib64/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
      File "/usr/lib64/python3.6/runpy.py", line 85, in _run_code
       exec(code, run_globals)
      File "/home/glenn/TWRP-device-tree-generator/twrpdtgen/__main__.py", line 6, in <module>
        main()
      File "/home/glenn/TWRP-device-tree-generator/twrpdtgen/twrp_dt_gen.py", line 56, in main
        build_prop = BuildPropReader(aik_ramdisk_path / "default.prop")
      File "/home/glenn/TWRP-device-tree-generator/twrpdtgen/info_extractors/buildprop.py", line 46, in __init__
       self._content = self._filename.read_text()
      File "/usr/lib64/python3.6/pathlib.py", line 1196, in read_text
       with self.open(mode='r', encoding=encoding, errors=errors) as f:
      File "/usr/lib64/python3.6/pathlib.py", line 1183, in open
       opener=self._opener)
      File "/usr/lib64/python3.6/pathlib.py", line 1037, in _opener
        return self._accessor.open(self, flags, mode)
      File "/usr/lib64/python3.6/pathlib.py", line 387, in wrapped
        return strfunc(str(pathobj), *args)
      FileNotFoundError: [Errno 2] No such file or directory: '/home/glenn/TWRP-device-tree-generator/extract/ramdisk   /default.prop'

Is the problem with my Python version (3.6.10), or possibly with the zip file that I am using? 
Thanks for taking the time to read this, as well as answering if you choose to do so. Kindest regards. Glenn

SebaUbuntu, 2020-10-09:
You need a recovery or a boot image (if the device is A/B)
zip format isn't supported

portsample, 2020-10-09:
Thank you for your response. Using boot.img met with some success. 
/Omni/out/target/product/device/ only contains clean_steps.mk and previous_build_config.mk, no recovery.img
Some of the messages during the build are:
...
"Running kati to generate build-omni_XP8800.ninja...
/external/f/software/minTWRPOmni/out/build-omni_XP8800.ninja is missing, regenerating...
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=16.1.0
TARGET_PRODUCT=omni_XP8800
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.12.14-lp151.28.67-default-x86_64-with-glibc2.2.5
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47F
OUT_DIR=/external/f/software/minTWRPOmni/out
============================================
Checking build tools versions...
including ./bionic/Android.mk ...
including ./bootable/recovery/Android.mk ...
bootable/recovery/prebuilt/Android.mk:600: Magisk prebuilt tools not found!
bootable/recovery/prebuilt/Android.mk:601: Please place https://github.com/TeamWin/external_magisk-prebuilt
bootable/recovery/prebuilt/Android.mk:602: into external/magisk-prebuilt
bootable/recovery/prebuilt/Android.mk:603: *** magiskboot prebuilts not present; exiting.
make: *** [build/core/ninja.mk:173: /external/f/software/minTWRPOmni/out/build-omni_XP8800.ninja] Error 1

#### make failed to build some targets (23 seconds) ####"

Used a boot.img from when Magisk was installed, then tried another (after doing "make clean") that was stock before Magisk and I still get the Magisk error. 
Interested in your thoughts on this. 
Best and highest regards to you for your work on this.

SebaUbuntu, 2020-10-09:
```
Please place https://github.com/TeamWin/external_magisk-prebuilt
into external/magisk-prebuilt
```

The build system literally said what's the issue

portsample, 2020-10-09:
Okay, that is a lot better. 
I am getting the following messages with no recovery.img file built. 

After "lunch omni_XP8800-eng" I get,

dfghom:/external/f/software/Omni # lunch omni_XP8800-eng
WARNING: device/sonimtech/XP8800/omni.dependencies file not found

============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=16.1.0
TARGET_PRODUCT=omni_XP8800
TARGET_BUILD_VARIANT=eng
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.12.14-lp151.28.67-default-x86_64-with-glibc2.2.5
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47F
OUT_DIR=/external/f/software/minTWRPOmni/out
============================================

Followed by "make recoveryimage" I get,

...
Running kati to generate build-omni_XP8800.ninja...
/external/f/software/minTWRPOmni/out/build-omni_XP8800.ninja is missing, regenerating...
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=16.1.0
TARGET_PRODUCT=omni_XP8800
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_BUILD_APPS=
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-4.12.14-lp151.28.67-default-x86_64-with-glibc2.2.5
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=NJH47F
OUT_DIR=/external/f/software/minTWRPOmni/out
============================================
Checking build tools versions...
including ./bionic/Android.mk ...
including ./bootable/recovery/Android.mk ...
including ./build/libs/host/Android.mk ...
including ./build/target/board/Android.mk ...
including ./build/target/product/security/Android.mk ...
including ./build/tools/Android.mk ...
including ./device/sonimtech/XP8800/Android.mk ...
[... 106 lines cut ...]
build/core/base_rules.mk:407: warning: ignoring old commands for target `/recovery_component_test'
build/core/base_rules.mk:407: warning: overriding commands for target `/minadbd_test'
build/core/base_rules.mk:407: warning: ignoring old commands for target `/minadbd_test'
bootable/recovery/toolbox/Android.mk:302: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/recovery/root/sbin/start'
bootable/recovery/toolbox/Android.mk:302: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/recovery/root/sbin/start'
bootable/recovery/toolbox/Android.mk:302: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/recovery/root/sbin/stop'
bootable/recovery/toolbox/Android.mk:302: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/recovery/root/sbin/stop'
build/core/base_rules.mk:319: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/system/xbin/unzip'
./external/busybox/Android.mk:172: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/system/xbin/unzip'
build/core/base_rules.mk:319: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/root/res/images/charger/battery_fail.png'
build/core/base_rules.mk:319: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/root/res/images/charger/battery_fail.png'
build/core/base_rules.mk:319: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/root/res/images/charger/battery_scale.png'
build/core/base_rules.mk:319: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/root/res/images/charger/battery_scale.png'
build/core/Makefile:1112: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/ramdisk-recovery.cpio'
build/core/Makefile:1068: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/ramdisk-recovery.cpio'
build/core/Makefile:1118: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/ramdisk-recovery.img'
build/core/Makefile:1074: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/ramdisk-recovery.img'
build/core/Makefile:1124: warning: overriding commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/recovery.img'
build/core/Makefile:1079: warning: ignoring old commands for target `/external/f/software/minTWRPOmni/out/target/product/XP8800/recovery.img'
Starting build with ninja
ninja: Entering directory `.'
ninja: error: '/external/f/software/minTWRPOmni/out/target/product/XP8800/system/xbin/zip', needed by '/external/f/software/minTWRPOmni/out/target/product/XP8800/obj_arm/SHARED_LIBRARIES/libbmlutils_intermediates/teamwin', missing and no known rule to make it
make: *** [build/core/ninja.mk:158: ninja_wrapper] Error 1

#### make failed to build some targets (33 seconds) ####

SebaUbuntu, 2020-10-11:
- Please use del.dog next time
- This seems your environment's issue

portsample, 2020-10-13:
Twrpdtgen and environment worked fine using a "test" device with a Lineage recovery.img. 
However "target" device has A/B partitions with no recovery.img partition on the device. 
How to progress with the A/B target device?
Appreciate your time and efforts. Thanks.

SebaUbuntu, 2020-10-13:
`mka bootimage` instead of `mka recoveryimage` should work

portsample, 2020-10-13:
Yes, that worked. 
Is "boot.img" install same as "recovery.img" install per instructions here: 
https://github.com/SebaUbuntu/TWRP-device-tree-generator/wiki/5.-Boot-the-recovery-image? 
Thanks for your work on this.

SebaUbuntu, 2020-10-13:
I think you must flash it as boot image and boot it with vol+ and power, you can't boot it with fastboot boot

closed 2020-10-13 by portsample

ChapayTv, 2020-10-18:
Здравствуйте, я не совсем понял что нужно сделать в третьем пункте, python3 -m twrpdtgen <path to the recovery image> после ввода этой команды выдаёт ошибку. У меня устройство Xiaomi A/B нет. Recovery.img я поместил в главную папку скрипта. Подскажите пожалуйста как мне нужно точно ввести команду в третьем пункте Wiki/ Заранее спасибо за ваш ответ

SebaUbuntu, 2020-10-18:
Please use English