twrpdtgen/twrpdtgen #16: Ошибка Phuton

ChapayTv, 2020-10-18:
Не могу понять почему у меня такая ошибка?

D:\TWRP-device-tree-generator-master>python -m twrpdtgen D:\Users\Chapay\Desktop\recovery.img
Traceback (most recent call last):
  File "C:\Python\lib\site-packages\git\__init__.py", line 83, in <module>
    refresh()
  File "C:\Python\lib\site-packages\git\__init__.py", line 73, in refresh
    if not Git.refresh(path=path):
  File "C:\Python\lib\site-packages\git\cmd.py", line 278, in refresh
    raise ImportError(err)
ImportError: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\TWRP-device-tree-generator-master\twrpdtgen\__main__.py", line 3, in <module>
    from twrpdtgen.twrp_dt_gen import main
  File "D:\TWRP-device-tree-generator-master\twrpdtgen\twrp_dt_gen.py", line 9, in <module>
    from git import Repo
  File "C:\Python\lib\site-packages\git\__init__.py", line 85, in <module>
    raise ImportError('Failed to initialize: {0}'.format(exc)) from exc
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

closed 2020-10-19 by SebaUbuntu

ChapayTv, 2020-10-20:
А где ответ?

SebaUbuntu, 2020-10-20:
English only