twrpdtgen/twrpdtgen #17: Successful build(s), TWRP not booting on device

portsample, 2020-10-19:
Both Twrpdtgen and Omni ROM TWRP builds went fine. Tried loading boot.img via fastboot and also using Qualcomm emergency download system (EDL) to flash the image. In addition, I did an unrooted build from a completely stock boot.img, as well as a build using a Magisk patched boot.img that was rooted and tried those using fastboot and EDL as well. Note that this device has A/B slots. Any suggestions? 
Thanks Sebastian for all of your hard work on this impressive project, as well as for your patience helping me with my earlier questions. Best regards.

SebaUbuntu, 2020-10-20:
Device?
Device tree link?
Stock recovery image?
Build recovery image?

portsample, 2020-10-23:
Device is Sonim XP8. Used boot.img. TWRPdtg /working/vendor is here: 
https://drive.google.com/file/d/1UZcYg7SlLTRJ7kXE7ZFY0hncK3ndNf3t
Thanks.

SebaUbuntu, 2020-10-27:
Access requested

portsample, 2020-11-21:
Successfully created boot.img after running TWRP-device-tree-generator output through Omni-rom for 8.1. 
Installed this image on BOTH boot_a and boot_b on the device. Device booted to Android OS no problem. 
Entered "ADB reboot recovery" in terminal, pressed enter, am now stuck at device splash screen. 
Is TWRP installed? How do I open it? So close and so far...thanks.

SebaUbuntu, 2020-11-30:
https://github.com/SebaUbuntu/TWRP-device-tree-generator/commit/acd68d10f9e50ec01c19a28fe1acead6387664a6
A possible fix has been pushed, needs testing

portsample, 2020-12-01:
TWRPdtg device tree build went fine. 
OmniTWRP boot.img build went fine.
Wrote boot.img over boot_a AND boot_b using Qualcomm EDL. 
Device booted into Android. Sent "adb reboot recovery". 
Device rebooted and stalled at bios before OS (or recovery) boot.
Next, overwrote boot_a with stock boot image, did "adb reboot recovery",
stalled at bios as before. Suggestions? Thanks again.

SebaUbuntu, 2020-12-01:
Could you please also send your boot image?

portsample, 2020-12-01:
> Could you please also send your boot image?

Both stock_boot and TWRP_boot images are here, 
https://drive.google.com/file/d/1SK15NSrzGp9VYL6CljLALUddGUvhVX2W/view?usp=sharing

portsample, 2020-12-06:
PPS...bootloader is unlocked. Thanks.

portsample, 2020-12-11:
Any idea what the issue might be Sebastian? Thanks.

yukosky, 2021-02-01:
> Any idea what the issue might be Sebastian? Thanks.

Which device?

portsample, 2021-02-01:
Sonim XP8800

SebaUbuntu, 2021-10-05:
Closing for inactivity

closed 2021-10-05 by SebaUbuntu