twrpdtgen/twrpdtgen #18: dtbo image check use wrong filename

yyjdelete, 2020-10-20:
See `xxx-recovery_dtbo` in `extract/split_img`, but seems python code checks `xxx-dtbo` and unable to found dtbo

Seems AIK also expected `-recovery_dtbo` in repackimg
https://github.com/SebaUbuntu/AIK-Linux-mirror/blob/master/repackimg.sh#L186-L189
https://github.com/SebaUbuntu/AIK-Windows-mirror/blob/master/repackimg.bat#L179-L181

https://github.com/SebaUbuntu/TWRP-device-tree-generator/blob/master/twrpdtgen/twrp_dt_gen.py#L83
https://github.com/SebaUbuntu/TWRP-device-tree-generator/blob/master/twrpdtgen/info_extractors/recovery_image.py#L26

SebaUbuntu, 2020-10-21:
oops
Yes, it needs to be fixed
Let me push a fix

SebaUbuntu, 2020-11-30:
Sorry for the delay, 3d30094f3b1cefa11aa829992c1060b2952abca6 should fix it

closed 2020-11-30 by SebaUbuntu