twrpdtgen/twrpdtgen #20: Error when building TWRP

AliceGrey, 2020-10-21:
```
FAILED: ninja: '/home/user/Dev/twrp/out/target/product/foles/obj/STATIC_LIBRARIES/bootctrl.sm6150_intermediates/export_includes', needed by '/home/user/Dev/twrp/out/target/product/foles/obj/SHARED_LIBRARIES/android.hardware.boot@1.0-impl-wrapper.recovery_intermediates/import_includes', missing and no known rule to make it
11:44:34 ninja failed with: exit status 1

```

Device Info:
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=16.1.0
TARGET_PRODUCT=omni_foles
TARGET_BUILD_VARIANT=eng
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-5.9.1-arch1-1-x86_64-Arch-Linux
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=QQ3A.200805.001
OUT_DIR=/home/user/Dev/twrp/out
ROM_BUILDTYPE=HOMEMADE

AliceGrey, 2020-10-21:
Device Tree: https://github.com/AliceGrey/twrp_device_Motorola_foles

SebaUbuntu, 2020-11-30:
You're building TWRP based on Android 7.1?

Hydrox6, 2020-12-01:
Having the same exact same issue trying to build TWRP for Oneplus Nord N10 on Android/TWRP 10

[Generated Device Tree](https://github.com/Hydrox6/minimal_twrp_tree_OnePlusN10)
[Device Info](https://gist.github.com/Hydrox6/90fca2d18121a432480503d50b6415ee)

Error line:
```
FAILED: ninja: '/mnt/media/Android/twrp-omni/out/target/product/OnePlusN10/obj/STATIC_LIBRARIES/bootctrl.lito_intermediates/export_includes', needed by '/mnt/media/Android/twrp-omni/out/target/product/OnePlusN10/obj/SHARED_LIBRARIES/android.hardware.boot@1.0-impl-wrapper.recovery_intermediates/import_includes', missing and no known rule to make it
```

bluehomewu, 2021-02-11:
> You're building TWRP based on Android 7.1?

Get the same error code for POCO M3.
When building

SebaUbuntu, 2021-10-05:
Closing for inactivity

closed 2021-10-05 by SebaUbuntu