twrpdtgen/twrpdtgen #21: Can unpack if magic and androidbootimg.magic are in windows langage

Elenedeath, 2020-10-26:
Hello i post you an issue and how to resolve it. i investigate and see what the problem and how to correct.

I clone your repository on windows and use py -m twrpdtgen TWRP331-BV9900_Android9-2019.img
[image]

After this i manually catch AIK from your repo and unzip in folder extract and try like this, oh nice it's working.

i compare with WinMerge original AIK and my github zip from your repo and see this
[image]

Due to this i search what the difference and all this file have something common, all are encoded in windows where other files are encoded on unix.

[image]

Simple fix and the really blocking file are 'androidbootimg.magic' and 'magic' need to changed to unix encoding and after this work fine.

This is the error when 'magic' file are unix encode but 'androidbootimg.magic' file are windows
[image]

This is the error when 'magic' file are windows encode but 'androidbootimg.magic' file are unix
[image]

SebaUbuntu, 2020-10-27:
AIK-Windows-mirror has been force pushed with both files with LF line endings, please retry

Elenedeath, 2020-10-27:
i try and didn't change, he changed previous files to CR+LF

closed 2020-10-30 by Elenedeath