twrpdtgen/twrpdtgen #22: Script should ask for build.prop values manually if it can't find them.

zacharee, 2020-11-11:
I attempted to use this script on a recovery image from the Realme V5 (shipped with Android 10). The recovery ramdisk does have a default.prop, but it's a symlink to another file in the system partition, and the script can't read from it.

Instead of raising an exception, it would be helpful if the script could ask the user if it can't find the default.prop or any values in an existing default.prop.

SebaUbuntu, 2020-11-12:
Could you please send your recovery image? So I can see if a build.prop is present elsewhere

zacharee, 2020-11-12:
https://drive.google.com/file/d/11YuLmvdcqS4htiXpSnTNad0MpcB1v9ke/view?usp=drivesdk

rokibhasansagar, 2020-11-28:
@SebaUbuntu, Some recoveries have `prop.default` as the original properties file with a symlink of `default.prop` inside the ramdisk.
This currently results in `FileNotFoundError: [Errno 2] No such file or directory: '/home/TWRP-device-tree-generator/extract/ramdisk/default.prop'` in some cases.

Add another method to find the __prop.default__ file if __default.prop__ is unavailable.

darkkingredian, 2020-11-29:
i have a same type of recovery (redmi 9a) but device tree was generated successfully

SebaUbuntu, 2021-10-05:
Closing for inactivity

closed 2021-10-05 by SebaUbuntu