twrpdtgen/twrpdtgen #23: Check support for dynamic partitions devices

SebaUbuntu, 2020-11-30:
i have a same type of recovery (redmi 9a) but device tree was generated successfully 
but there is no system partition location [device tree](https://github.com/darkkingredian/9atwrptest/blob/main/recovery.fstab)

_Originally posted by @darkkingredian in https://github.com/SebaUbuntu/TWRP-device-tree-generator/issues/22#issuecomment-735402772_

SebaUbuntu, 2020-11-30:
@darkkingredian could you please send your stock recovery image?

darkkingredian, 2020-12-01:
```
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ashraful/Music/TWRP-device-tree-generatorad/twrpdtgen/__main__.py", line 6, in <module>
    main()
  File "/home/ashraful/Music/TWRP-device-tree-generatorad/twrpdtgen/twrp_dt_gen.py", line 58, in main
    build_prop = BuildPropReader(recovery_image_info.buildprop)
  File "/home/ashraful/Music/TWRP-device-tree-generatorad/twrpdtgen/info_extractors/buildprop.py", line 30, in __init__
    self.codename = self.get_prop(DEVICE_CODENAME_RE, "codename")
  File "/home/ashraful/Music/TWRP-device-tree-generatorad/twrpdtgen/info_extractors/buildprop.py", line 47, in get_prop
    raise AssertionError(f"Device {error} could not be found in build.prop")
AssertionError: Device codename could not be found in build.prop

```
i don"t remember how it succeed yesterterday but this happened today
the recovery img [https://drive.google.com/file/d/1JmCVMa98iJtl0jJfySZbZ0sLreT5kVOW/view?usp=sharing](url)

SebaUbuntu, 2020-12-01:
> ```
> Traceback (most recent call last):
>   File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
>     return _run_code(code, main_globals, None,
>   File "/usr/lib/python3.8/runpy.py", [...]

This has been fixed with the last commit, I'm now fixing logical partition fstab

SebaUbuntu, 2020-12-01:
> i have a same type of recovery (redmi 9a) but device tree was generated successfully
> but there is no system partition location [device tree](https://github.com/darkkingredian/9atwrptest/blob/main/recovery.fstab)

Fixed with latest commit

closed 2020-12-01 by SebaUbuntu

darkkingredian, 2020-12-01:
thanks man it worked