twrpdtgen/twrpdtgen #24: erro

andyhang1980, 2020-12-02:
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.8/runpy.py", line 144, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/home/zfz/TWRP-device-tree-generator/twrpdtgen/__init__.py", line 2, in <module>
    from jinja2 import Environment, FileSystemLoader
ModuleNotFoundError: No module named 'jinja2'

SebaUbuntu, 2020-12-02:
You haven't did `pip3 install .`

SebaUbuntu, 2020-12-02:
Reopen it if there are issues

closed 2020-12-02 by SebaUbuntu