twrpdtgen/twrpdtgen #28: Succesful Build, Bootloop, How do I debug?

diabl0w, 2020-12-15:
Hi, thanks so much for this twrp config generator. Everything went pretty smooth for the most part in getting an image to build. I know this isn't the TWRP repo, but this is the only active place I can find about twrp building issues. 

I have the image built and can succesfully flash it, but i just end up in a bootloop that eventually just boots to system after 2-3 reboot loops. Can you just point me to somewhere that I can try to debug this, somewhere with logs? I tried looking in /cache/recovery/ in a booted system but there is nothing new that gets created there.

If it helps, I am building for an Android 9 system. It is NOT A/B, just normal. It has a boot.img and a recovery.img, but the recovery.img is the one that holds the ramdisk. I am using twrp-device-tree-generator on the recovery.img and then building twrp using the "mka recoveryimg" command

diabl0w, 2020-12-15:
edit: I noticed there a lot of folders like "cache" "vendor" "system_root" that are missing from the twrp recovery.img as compared to the stock recovery.img.... could this be why? how could I insert these into the twrp img?

SebaUbuntu, 2020-12-22:
Send your recovery image and your generated device tree

tiann, 2021-04-09:
Maybe it is because https://github.com/SebaUbuntu/TWRP-device-tree-generator/pull/45, this bug will treat non A/B devices as A/B device

SebaUbuntu, 2021-10-05:
Closing for inactivity

closed 2021-10-05 by SebaUbuntu

baejeongwan, 2024-01-17:
How did you fixed it? I'm having same issue.

Uploading compressed recovery.img
[Uploading recovery.zip…]()