twrpdtgen/twrpdtgen #32: posixpath error

steve4655, 2020-12-27:
root@LAPTOP-9Q27PVAC:/home/steve/TWRP-device-tree-generator# python3 -m twrpdtgen /home/steve/TWRP/recovery.img
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/steve/TWRP-device-tree-generator/twrpdtgen/__init__.py", line 10, in <module>
    jinja_env = Environment(loader=FileSystemLoader(current_path / 'templates'),
  File "/usr/lib/python3/dist-packages/jinja2/loaders.py", line 163, in __init__
    self.searchpath = list(searchpath)
TypeError: 'PosixPath' object is not iterable

steve4655, 2020-12-27:
anyone know how to fix

SebaUbuntu, 2020-12-27:
Have you fully cloned the script with Git?

CyberRoamer, 2021-01-02:
Idk if it works or not but modifying __init__.py to the below helped me past this error in ubuntu 18.04 but I was unable to completely build twrp using the output I got. Then I tried Ubuntu 16.04 to check if I can build TWRP is showed another error.

module_path = str(Path(__file__).parent)
current_path = Path(os.getcwd())

jinja_env = Environment(loader=FileSystemLoader(module_path , 'templates'),
                        autoescape=True, trim_blocks=True, lstrip_blocks=True)

CyberRoamer, 2021-01-02:
Idk if it works or not but modifying` __init__.py ` in twrpdtgen to the below helped me past this error in ubuntu 18.04 but I was unable to completely build twrp using the output I got. 
```
module_path = str(Path(__file__).parent)
current_path =Path(os.getcwd())

jinja_env = Environment(loader=FileSystemLoader(module_path , 'templates'),
                        autoescape=True, trim_blocks=True, lstrip_blocks=True)

```
Then I tried Ubuntu 16.04 to check if that can build TWRP where I was unable to get the same results. Sadly I forgot to note down python version 18.04 was using. But I was able to get past the error by modifying the below in the same file

```
module_path = '<<complete path till templates folder in twrpdtgen>>'     ##eg. '/home/<user>/twrptreebuilder/twrpdtgen/templates'
current_path =Path(os.getcwd())

jinja_env = Environment(loader=FileSystemLoader(module_path),
                        autoescape=True, trim_blocks=True, lstrip_blocks=True)

```

SebaUbuntu, 2021-01-03:
I pushed some commits, could you please retry?

Hadenix, 2021-01-03:
> I pushed some commits, could you please retry?

```
 python twrpdtgen -h
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "twrpdtgen/__main__.py", line 5, in <module>
    from twrpdtgen import __version__ as version, current_path
ModuleNotFoundError: No module named 'twrpdtgen'
```

SebaUbuntu, 2021-01-03:
python3 -m twrpdtgen recoveryimage

steve4655, 2021-01-05:
done sar it is fixed
:)

steve4655, 2021-01-05:
> Have you fully cloned the script with Git?

yea and it is fixed now 
BTW great work bro :)

closed 2021-01-05 by steve4655