twrpdtgen/twrpdtgen #35: Twrpdtgen doesnt install corrrectly

Exynos-nigg, 2021-01-07:
uh, so a kinda weird and strange issue, the thing is i have installed twrpdtgen but when i execute `twrpdtgen -h` it gives me the output `-bash: twrpdtgen: command not found`

**Commands executed:**
`bruh-buildbot@bruh-buildbot:~/android$ pip3 install twrpdtgen
Collecting twrpdtgen
  Downloading twrpdtgen-1.2.2-py3-none-any.whl (16 kB)
Requirement already satisfied: GitPython<4.0.0,>=3.1.11 in /usr/local/lib/python3.8/dist-packages (from twrpdtgen) (3.1.12)
Requirement already satisfied: Jinja2<3.0.0,>=2.11.2 in /usr/local/lib/python3.8/dist-packages (from twrpdtgen) (2.11.2)
Requirement already satisfied: gitdb<5,>=4.0.1 in /usr/local/lib/python3.8/dist-packages (from GitPython<4.0.0,>=3.1.11->twrpdtgen) (4.0.5)
Requirement already satisfied: MarkupSafe>=0.23 in /usr/lib/python3/dist-packages (from Jinja2<3.0.0,>=2.11.2->twrpdtgen) (1.1.0)
Requirement already satisfied: smmap<4,>=3.0.1 in /usr/local/lib/python3.8/dist-packages (from gitdb<5,>=4.0.1->GitPython<4.0.0,>=3.1.11->twrpdtgen) (3.0.4)
Installing collected packages: twrpdtgen
Successfully installed twrpdtgen-1.2.2
bruh-buildbot@bruh-buildbot:~/android$ twrpdtgen -h
-bash: twrpdtgen: command not found`

**Note**

- just to see if it was just a "fluke", i tried installing on my local pc (the above output was done on a gcp vm instance) and it was the same as above, output as follows: 
`bruh@bruh-ROG-Strix-b10esx:~$ twrpdtgen -h
twrpdtgen: command not found`

- i have tried installing it with `sudo pip install` and `sudo pip3 install` yet it yields the same output

- i do have cpio installed and configured.

any help would be appreciated!

SebaUbuntu, 2021-01-07:
python3 -m twrpdtgen ...

SebaUbuntu, 2021-01-07:
Situation improved with commit 81a9740a93679029b600699a462f3d0bda902d99

closed 2021-01-07 by SebaUbuntu