twrpdtgen/twrpdtgen #36: Crashes when cloning AIK

AliceGrey, 2021-01-17:
Latest version 1.2.3 crashes when cloning AIK

`TWRP device tree generator
Version 1.2.3

[aik_manager.py:41 INFO] __init__: Cloning AIK...
[cmd.py:718 DEBUG] execute: Popen(['git', 'clone', '-v', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/tmp/tmpdpjwdzer'], cwd=/home/alice/Desktop/Bramble, universal_newlines=True, shell=None, istream=None)
[base.py:972 DEBUG] _clone: Cmd(['git', 'clone', '-v', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/tmp/tmpdpjwdzer'])'s unused stdout: 
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/alice/.local/lib/python3.9/site-packages/twrpdtgen/__main__.py", line 28, in <module>
    dt = generate_device_tree(args.recovery_image, args.output, no_git=args.no_git, keep_aik=args.keep_aik)
  File "/home/alice/.local/lib/python3.9/site-packages/twrpdtgen/twrp_dt_gen.py", line 30, in generate_device_tree
    aik.extract(recovery_image)
  File "/home/alice/.local/lib/python3.9/site-packages/twrpdtgen/utils/aik_manager.py", line 64, in extract
    self.get_image_infos()
  File "/home/alice/.local/lib/python3.9/site-packages/twrpdtgen/utils/aik_manager.py", line 79, in get_image_infos
    self.base_address = self.read_recovery_file(self.get_extracted_info("base"))
  File "/home/alice/.local/lib/python3.9/site-packages/twrpdtgen/utils/aik_manager.py", line 108, in read_recovery_file
    return file.read_text().splitlines()[0]
  File "/usr/lib/python3.9/pathlib.py", line 1255, in read_text
    with self.open(mode='r', encoding=encoding, errors=errors) as f:
  File "/usr/lib/python3.9/pathlib.py", line 1241, in open
    return io.open(self, mode, buffering, encoding, errors, newline,
  File "/usr/lib/python3.9/pathlib.py", line 1109, in _opener
    return self._accessor.open(self, flags, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/tmpdpjwdzer/split_img/recovery.img-base'
`

SebaUbuntu, 2021-01-19:
Send your recovery image

AliceGrey, 2021-01-24:
@SebaUbuntu It's the stock Pixel 4A 5G recovery image. It's available on Google's website for download.

SebaUbuntu, 2021-03-30:
There are now the necessary tools to debug it, please reopen the issue with proper logging if you are still interested

closed 2021-03-30 by SebaUbuntu