twrpdtgen/twrpdtgen #37: Crashes after "Cloning AIK"

Divkix, 2021-01-19:
Steps:
```
divkix@Inspiron5459:~/TWRP_Device_Generator$ python3 -m twrpdtgen boot.img
TWRP device tree generator
Version 1.2.3

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/divkix/.local/lib/python3.8/site-packages/twrpdtgen/__main__.py", line 28, in <module>
    dt = generate_device_tree(args.recovery_image, args.output, no_git=args.no_git, keep_aik=args.keep_aik)
  File "/home/divkix/.local/lib/python3.8/site-packages/twrpdtgen/twrp_dt_gen.py", line 34, in generate_device_tree
    raise AssertionError("Couldn't find any build.prop")
AssertionError: Couldn't find any build.prop
```

I'm using Asus ROG Phone 2 boot.img for testing and this error is thrown, have also used ROG Phone 3 boot.img, still same result.

SebaUbuntu, 2021-01-19:
Send your recovery/boot image

Divkix, 2021-01-19:
Here is the Rog 2 boot.img:
https://drive.google.com/file/d/1-RM7-4XA7tFssYQe9J0W91xvfGHNQMpG/view?usp=drivesdk

Hadenix, 2021-01-23:
> Send your recovery/boot image

no solution yet?

SebaUbuntu, 2021-01-23:
> > Send your recovery/boot image
> 
> no solution yet?

If you are talking about another device, please open another issue

Hadenix, 2021-01-23:
@SebaUbuntu i tried i got the same error

SebaUbuntu, 2021-01-23:
> @SebaUbuntu i tried i got the same error

With ROG 2?

SebaUbuntu, 2021-03-30:
There are now the necessary tools to debug it, please reopen the issue with proper logging if you are still interested

closed 2021-03-30 by SebaUbuntu