twrpdtgen/twrpdtgen #40: Samsung SM-T590 bricked, impossible to reach download mode

UDPSendToFailed, 2021-02-12:
I have tried to compile OrangeFox recovery using device tree generated by this script, it went fine. I flashed it, the recovery isn't even loading, just flashes the screen once and goes black. Adb is not available, and it doesn't matter which key combination I press, it goes straight to recovery again and again.

Any ideas on how to fix this? 

Here is the device tree I used: https://github.com/UDPSendToFailed/twrp_device_samsung_gta2xlwifi

durasame, 2021-02-12:
Wait for the battery to run out

SebaUbuntu, 2021-02-12:
I had a similar issue with j4primelte
The solution is:
- Let it discharge
- Once it powered off, connect the device to a PC with a USB cable while keeping Odin mode keys, in my case, Vol- (Volume Down)
- This time, DON'T CONFIRM ENTER ODIN MODE, just let it charge in warning screen
- Wait some hours, then reflash a working recovery with Odin

UDPSendToFailed, 2021-02-12:
Thanks guys, so there are no secret key combinations to force it to turn off. :/ Maybe I will go ahead with disassembling it and disconnecting the battery, because it would take a really long time to drain while the screen is off. Did the OrangeFox recovery cause this, or am I missing something?

SebaUbuntu, 2021-02-12:
> Did the OrangeFox recovery cause this, or am I missing something?
Yes, unfortunately making a custom recovery for Samsung devices may lead to this, and surely not having fastboot makes this stuff harder

> is there a less risky way to test if a recovery is working or not?
Not that I'm aware

UDPSendToFailed, 2021-02-12:
Thanks again, I will disassemble it tomorrow, disconnect the battery and report back.

closed 2021-02-12 by UDPSendToFailed

UDPSendToFailed, 2021-02-13:
Disconnecting the battery did the trick, I was able to reflash TWRP, although on USB power. When the battery was connected, it was booting to recovery regardless key combinations. 

[image]