twrpdtgen/twrpdtgen #48: Any solutions Sir

boedhack99, 2021-04-14:
$ python3 -m twrpdtgen recovery.img
TWRP device tree generator
Version 1.2.5

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/data/data/com.termux/files/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/data/data/com.termux/files/usr/lib/python3.9/site-packages/twrpdtgen/__main__.py", line 52, in <module>
    dt = DeviceTree(args.output, recovery_image=args.recovery_image,
  File "/data/data/com.termux/files/usr/lib/python3.9/site-packages/twrpdtgen/device_tree.py", line 49, in __init__
    aik.extract(recovery_image)
  File "/data/data/com.termux/files/usr/lib/python3.9/site-packages/twrpdtgen/utils/aik_manager.py", line 74, in extract                                    raise RuntimeError(f"AIK extraction failed, return code {returncode}")  RuntimeError: AIK extraction failed, return code 1

friendlymaninthisworld, 2021-04-25:
Maybe you have a a/b partition system? 
Try the boot.img, i guess

boedhack99, 2021-04-26:
Yes it is boot.img but i rename it to recovery.img

Robert3141, 2021-05-03:
I get the same error when trying to use boot.img on the firmware for my Poco F3 (launched with Android 11). I know my device has A/B partitioning and uses dynamic partitions so that might be the problem

boedhack99, 2021-05-03:
> I get the same error when trying to use boot.img on the firmware for my Poco F3 (launched with Android 11). I know my device has A/B partitioning and uses dynamic partitions so that might be the problem

Using unpackbootimg & mkbootimg latest from osm0sis for repacked,bcz f3 & rn10 same using hdrv3

Robert3141, 2021-05-03:
> > I get the same error when trying to use boot.img on the firmware for my Poco F3 (launched with Android 11). I know my device has A/B partitioning and uses dynamic partitions so that might be the problem
> 
> Using unpackbootimg & mkbootimg latest from osm0sis for repacked,bcz f3 & rn10 same using hdrv3

I'm confused. What do you mean?

closed 2021-05-16 by boedhack99