twrpdtgen/twrpdtgen #49: What's with "clang policy_define.c" ?

friendlymaninthisworld, 2021-04-24:
FAILED: /home/aqua/Twrp/out/soong/.intermediates/external/selinux/checkpolicy/checkpolicy/linux_x86_64/gen/lex/external/selinux/checkpolicy/policy_scan.c 
prebuilts/misc/linux-x86/flex/flex-2.5.39 -o/home/aqua/Twrp/out/soong/.intermediates/external/selinux/checkpolicy/checkpolicy/linux_x86_64/gen/lex/external/selinux/checkpolicy/policy_scan.c external/selinux/checkpolicy/policy_scan.l
flex-2.5.39: loadlocale.c:130: _nl_intern_locale_data: Assertion `cnt < (sizeof (_nl_value_type_LC_TIME) / sizeof (_nl_value_type_LC_TIME[0]))' failed.
Aborted (core dumped)
[ 20% 2326/11354] //external/selinux/checkpolicy:checkpolicy clang policy_define.c [linux]
ninja: build stopped: subcommand failed.
00:17:54 ninja failed with: exit status 1

#### failed to build some targets (05:31 (mm:ss)) ####

I have been following the wiki step by step however nothing I do can get over this problem. Anyone know the solution?

SebaUbuntu, 2021-04-24:
```
export LC_ALL=C
mka recoveryimage
```

friendlymaninthisworld, 2021-04-25:
AMAZING it's finally past 50% thanks, but I ran into a problem :

clang++.real: error: unable to execute command: Segmentation fault (core dumped)
clang++.real: error: clang frontend command failed due to signal (use -v to see invocation)
Android clang version 5.0.300080  (based on LLVM 5.0.300080)
Target: x86_64--linux-gnu
Thread model: posix
InstalledDir: prebuilts/clang/host/linux-x86/clang-4053586/bin
clang++.real: note: diagnostic msg: PLEASE submit a bug report to http://llvm.org/bugs/ and include the crash backtrace, preprocessed source, and associated run script.
clang++.real: note: diagnostic msg: 
********************

PLEASE ATTACH THE FOLLOWING FILES TO THE BUG REPORT:
Preprocessed source(s) and associated run script(s) are located at:
clang++.real: note: diagnostic msg: /tmp/AsmMatcherEmitter-a61650.cpp
clang++.real: note: diagnostic msg: /tmp/AsmMatcherEmitter-a61650.sh
clang++.real: note: diagnostic msg: 

********************
[ 62% 7079/11354] //frameworks/native/opengl/libs:libGLESv2.n...GLESv2.ndk/android_arm_armv7-a-neon_shared_core_18/gen/stub.c
ninja: build stopped: subcommand failed.
22:02:56 ninja failed with: exit status 1
Maybe it's a bug?

friendlymaninthisworld, 2021-04-25:
I'll open a new issue.

closed 2021-04-25 by friendlymaninthisworld