twrpdtgen/twrpdtgen #62: AssertionError while using the script.

Mr-khan007, 2021-07-17:
Trying this script to generate a dt for my device. But getting below error, and also Below is the link for my recovery..

raise AssertionError(f"Device {error} could not be found in build.prop")
AssertionError: Device arch could not be found in build.prop

[RECOVERY](https://true-mirror-69.gitsauraj.workers.dev/0:/C.08/C.08/recovery.img)

techyminati, 2021-07-18:
care sharing full logs? I guess , this device arch issue was fixed in commit https://github.com/SebaUbuntu/TWRP-device-tree-generator/commit/b38c6f07504cc9e4f35c9261d0f7e2bb4fa95815 for most devices, still full error logcat appreciated.

techyminati, 2021-07-18:
@Mr-khan007 Probably You're using older version of script, re clone it or git pull recent changes, I've just generated a DT from your given recoveryimage & pushed it here , https://github.com/MinatiScape/omni_device_alps_oppo6785 . @SebaUbuntu u can close this.

Mr-khan007, 2021-07-18:
@techyminati thanx for ur reply...

closed 2021-07-18 by Mr-khan007