twrpdtgen/twrpdtgen #83: AssertionError: Device codename could not be found in build.prop

Hadenix, 2021-09-15:
**Device:** _Huawei Y6 Prime 2018_
**Model:** _ATU-L31_

___
`python3 -m twrpdtgen --huawei --recovery_kernel RECOVERY_KERNEL --recovery_ramdisk RECOVERY_RAMDISK --recovery_vendor RECOVERY_VENDOR
TWRP device tree generator
Version 1.3.0

[INFO] Cloning AIK...
[INFO] Cloning AIK...
[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/hadenix/.local/lib/python3.8/site-packages/twrpdtgen/__main__.py", line 52, in <module>
    dt = DeviceTree(args.output, recovery_image=args.recovery_image,
  File "/home/hadenix/.local/lib/python3.8/site-packages/twrpdtgen/device_tree.py", line 60, in __init__
    self.build_prop_reader = BuildPropReader(self.build_prop)
  File "/home/hadenix/.local/lib/python3.8/site-packages/twrpdtgen/info_extractors/buildprop.py", line 28, in __init__
    self.codename = self.get_prop(DEVICE_CODENAME, "codename")
  File "/home/hadenix/.local/lib/python3.8/site-packages/twrpdtgen/info_extractors/buildprop.py", line 61, in get_prop
    raise AssertionError(f"Device {error} could not be found in build.prop")
AssertionError: Device codename could not be found in build.prop`

Hadenix, 2021-09-15:
You can try to add some standard names. To use this as the default if the script doesn't find the values you want.

_I did not have:_
**DEVICE_CODENAME**
**DEVICE_BRAND**
**DEVICE_MODEL**

_in my case:_ I modified buildprop.py by adding "ro.build.user" to each:
**DEVICE_CODENAME** = + ["ro.build.user"]
**DEVICE_BRAND** = + ["ro.build.user"]
**DEVICE_MODEL** = + ["ro.build.user"]

SebaUbuntu, 2021-09-15:
That's not correct, ro.build.user is the build machine user

Hadenix, 2021-09-19:
> That's not correct, ro.build.user is the build machine user

this is an option to build a tree. later you can already change to your names

SebaUbuntu, 2021-09-19:
No, I'll never do something like this

SebaUbuntu, 2021-10-05:
Huawei support has been dropped

closed 2021-10-05 by SebaUbuntu