twrpdtgen/twrpdtgen #86: How to deal with this problem

58773622, 2021-10-02:
root@PC:/home/pc/tee# python3 -m twrpdtgen recovery.img
TWRP device tree generator
Version 2.0.1

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/pc/tee/twrpdtgen/__main__.py", line 6, in <module>
    main()
  File "/home/pc/tee/twrpdtgen/main.py", line 31, in main
    device_tree = DeviceTree(image=args.image, debug=args.debug)
  File "/home/pc/tee/twrpdtgen/devicetree.py", line 62, in __init__
    self.deviceinfo = DeviceInfo(self.buildprop)
  File "/home/pc/tee/twrpdtgen/utils/deviceinfo.py", line 74, in __init__
    self.codename = self.get_prop(DEVICE_CODENAME)
  File "/home/pc/tee/twrpdtgen/utils/deviceinfo.py", line 93, in get_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.device could not be found in build.prop

SebaUbuntu, 2021-11-03:
> root@PC:/home/pc/tee# python3 -m twrpdtgen recovery.img TWRP device tree generator Version 2.0.1
> 
> Traceback (most recent call last): File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main return _run_code(code, main_globals, None, File "/usr/lib/python3.8/runpy.py", line 87, in _run_code exec(code, run_globals) File "/home/pc/tee/twrpdtgen/**main**.py", line 6, in main() File "/home/pc/tee/twrpdtgen/main.py", line 31, in main device_tree = DeviceTree(image=args.image, debug=args.debug) File "/home/pc/tee/twrpdtgen/devicetree.py", line 62, in **init** self.deviceinfo = DeviceInfo(self.buildprop) File "/home/pc/tee/twrpdtgen/utils/deviceinfo.py", line 74, in **init** self.codename = self.get_prop(DEVICE_CODENAME) File "/home/pc/tee/twrpdtgen/utils/deviceinfo.py", line 93, in get_prop raise AssertionError(f'Property {props[0]} could not be found in build.prop') AssertionError: Property ro.product.device could not be found in build.prop

Your image doesn't comtain that prop, there's nothing I can do about it

closed 2021-11-03 by SebaUbuntu

58773622, 2021-11-03:
Thank you. I'll study it