twrpdtgen/twrpdtgen #91: About repurchasing AIK on every run

xxroot, 2021-10-19:
Hello brother

As for the problem that AIK needs to be repurchased every time it runs, can we remove this problem? It takes a long time to generate the device tree every time. Can we change it to the first run, repurchase AIK, and no longer repurchase AIK after running

SebaUbuntu, 2021-10-19:
Repurchase?

SebaUbuntu, 2021-11-03:
> Android Image Kitchen Every time you run it, you have to re git clone this unpacking tool. Can you put this unpacking tool in a folder instead of git clone every time

Sorry for the late reply
Unfortunately I can't do it because cleaning up extraction files can sometimes fail, I prefer just recloning everything from scratch

closed 2021-11-03 by SebaUbuntu