twrpdtgen/twrpdtgen #92: Black screen after booting

danieleremin, 2021-11-04:
I flashed the compiled twrp for my phone (SM-J337U) and it booted into a black screen.

faberto, 2021-11-05:
Same for miPad 5 ( nabu)

SebaUbuntu, 2021-11-05:
Try to remove https://github.com/twrpdtgen/android_device_10or_G/blob/G-user-8.1.0-OPM1.171019.019-10or_G_V1_0_82-release-keys/BoardConfig.mk#L68

danieleremin, 2021-11-06:
> Try to remove https://github.com/twrpdtgen/android_device_10or_G/blob/G-user-8.1.0-OPM1.171019.019-10or_G_V1_0_82-release-keys/BoardConfig.mk#L68

Remove or set to false?

SebaUbuntu, 2021-11-06:
> > Try to remove https://github.com/twrpdtgen/android_device_10or_G/blob/G-user-8.1.0-OPM1.171019.019-10or_G_V1_0_82-release-keys/BoardConfig.mk#L68
> 
> Remove or set to false?

It's the same

danieleremin, 2021-11-06:
Same issue.

danieleremin, 2021-11-28:
Could this be related to vbmeta?

SebaUbuntu, 2022-09-28:
Closing for inactivity, feel free to open it back if the issue persist with the latest version

closed 2022-09-28 by SebaUbuntu