twrpdtgen/twrpdtgen #100: No MRVL support

TriDiscord, 2021-12-22:
MRVL devices are not really supported and most TWRP builds made with the device trees don't work at ALL.

SebaUbuntu, 2022-01-07:
An example?

TriDiscord, 2022-01-18:
Let me give some BoardConfig flags as examples:
`MRVL_ION`
`BOARD_USES_MRVL_HARDWARE`
`MRVL_INTERFACE_ANIMATION`
`MRVL_LAUNCH_DMS_IN_SURFACEFLINGER`

There *are* some other flags for Marvell chips but these are the most critical ones. `TARGET_BOARD_PLATFORM` is detected correctly.

SebaUbuntu, 2022-01-18:
After a search with cs.github.com I've concluded that none of these flags does something, there's no match anywhere except a few device trees
If you can find an AOSP component which uses these flags let me know, but until now, mrvl support is there

closed 2022-01-18 by SebaUbuntu

TriDiscord, 2022-01-19:
I can't really find anything about it on AOSP, _however_, I do see some official TWRP repositories use them.

[4 official devices](https://github.com/search?q=org%3ATeamWin+MRVL_ION&type=code)

SebaUbuntu, 2022-01-19:
Yes, but they're useless since no other repo reference them
y'know flags are supposed to set something, but if they aren't used they're useless

SebaUbuntu, 2022-01-19:
This equals to:
FInd any repo that reference these flags outside device trees

TriDiscord, 2022-01-19:
All Marvell PXA-series (Armanda) chips use Vivante.

Final USED flags:
```
# MRVL support
TARGET_FORCE_CPU_UPLOAD := true
BOARD_USE_VIVANTE_GRALLOC := true
HDMI_SUPPORT_3D := true
USE_OPENGL_RENDERER := true
BOARD_HAVE_PIXEL_FORMAT_INFO := true
```
Will be opening PR.

SebaUbuntu, 2022-01-19:
All of these are not used within bootable/recovery, thus they're useless when building a recovery