twrpdtgen/twrpdtgen #110: AssertionError: Couldn't find any build.prop

Nomz420, 2022-02-16:
```
sudo python3 -m twrpdtgen recovery.img -d
TWRP device tree generator
Version 2.1.0

[aikmanager.py:54 DEBUG] __init__: Cloning AIK...
[cmd.py:869 DEBUG] execute: Popen(['git', 'clone', '-v', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/media/hdd/extract'], cwd=/media/hdd, universal_newlines=True, shell=None, istream=None)
[base.py:1085 DEBUG] _clone: Cmd(['git', 'clone', '-v', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/media/hdd/extract'])'s unused stdout: 
[devicetree.py:51 DEBUG] __init__: Getting device infos...
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 6, in <module>
    main()
  File "/usr/lib/python3.10/site-packages/twrpdtgen/main.py", line 31, in main
    device_tree = DeviceTree(image=args.image, debug=args.debug)
  File "/usr/lib/python3.10/site-packages/twrpdtgen/devicetree.py", line 59, in __init__
    raise AssertionError("Couldn't find any build.prop")
AssertionError: Couldn't find any build.prop
```

recovery.img: https://nomz.ct8.pl/cdn/recovery.img

trkzmn, 2022-03-26:
The image you provided doesn't include any build.prop which is not a generic recovery ramdisk. This kind ramdisk is not supported by this tool.

SebaUbuntu, 2022-04-28:
This recovery doesn't have any working build.prop and prop.default points to /cache/runtime.prop, can't do anything about it

closed 2022-04-28 by SebaUbuntu

Poempoat, 2022-10-05:
You can use android image kitchen to unpack image and add build.prop at ramdisk from system.It works very well for me.