twrpdtgen/twrpdtgen #112: Cannot find fstab when trying to run create device tree with ELE-L29 recovery image.

j7b3y, 2022-03-08:
How can I solve this problem?
Recovery image:[https://drive.google.com/file/d/1bFFMbAplWFBS4ZET7029lUfVctjWSN9F/view?usp=sharing](https://drive.google.com/file/d/1bFFMbAplWFBS4ZET7029lUfVctjWSN9F/view?usp=sharing)
```
sudo python3 -m twrpdtgen '/home/j7b3y/Desktop/RECOVERY_RAMDIS.img'  -d
           
TWRP device tree generator
Version 2.1.0

[aikmanager.py:54 DEBUG] __init__: Cloning AIK...
[cmd.py:870 DEBUG] execute: Popen(['git', 'clone', '-v', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/home/j7b3y/extract'], cwd=/home/j7b3y, universal_newlines=True, shell=None, istream=None)
[base.py:1085 DEBUG] _clone: Cmd(['git', 'clone', '-v', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/home/j7b3y/extract'])'s unused stdout: 
[devicetree.py:51 DEBUG] __init__: Getting device infos...
[devicetree.py:61 DEBUG] __init__: Using <twrpdtgen.utils.buildprop.BuildProp object at 0x7f662c6faeb0> as build.prop
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/twrpdtgen/__main__.py", line 6, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/twrpdtgen/main.py", line 31, in main
    device_tree = DeviceTree(image=args.image, debug=args.debug)
  File "/usr/local/lib/python3.9/dist-packages/twrpdtgen/devicetree.py", line 81, in __init__
    raise AssertionError("fstab not found")
AssertionError: fstab not found
```

trkzmn, 2022-03-26:
The image you (huawei) provided doesn't include recovery fstab which is not a generic recovery ramdisk. This kind ramdisk is not supported by this tool.

j7b3y, 2022-03-26:
Okay

closed 2022-03-26 by j7b3y

acastillorobles77, 2022-07-14:
Hey I have the same issue with a P smart 2019 (POT-LX3) recovery ramdisk image