twrpdtgen/twrpdtgen #129: what should i do ??

Hostfl, 2022-07-21:
Android Image Kitchen - UnpackImg Script
by osm0sis @ xda-developers

Supplied image: recovery.img

Setting up work folders...

Unrecognized format.
Error!

Traceback (most recent call last):
  File "/data/data/com.termux/files/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/data/data/com.termux/files/usr/lib/python3.10/runpy.py", line 86, in _run_code                         exec(code, run_globals)                             File "/data/data/com.termux/files/usr/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()                                              File "/data/data/com.termux/files/usr/lib/python3.10/site-packages/twrpdtgen/main.py", line 37, in main       device_tree = DeviceTree(image=args.image)
  File "/data/data/com.termux/files/usr/lib/python3.10/site-packages/twrpdtgen/device_tree.py", line 53, in __init__
    self.image_info = self.aik_manager.unpackimg(image)
  File "/data/data/com.termux/files/usr/lib/python3.10/site-packages/sebaubuntu_libs/libaik/__init__.py", line 122, in unpackimg
    raise RuntimeError(f"AIK extraction failed, return code {returncode}")
RuntimeError: AIK extraction failed, return code 1

HelloDefender12, 2022-09-06:
Try to use python 3.8

HelloDefender12, 2022-09-18:
It won't work on termux

SebaUbuntu, 2022-09-28:
AIK issue, not related to twrpdtgen

closed 2022-09-28 by SebaUbuntu