twrpdtgen/twrpdtgen #143: AssertionError: Property ro.product.bootimage.device could not be found in build.prop

ghostranger047, 2023-01-07:
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/avenger047/.local/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/avenger047/.local/lib/python3.10/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
  File "/home/avenger047/.local/lib/python3.10/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
  File "/home/avenger047/.local/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 117, in __init__
    self.codename = self.get_first_prop(DEVICE_CODENAME)
  File "/home/avenger047/.local/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 177, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.bootimage.device could not be found in build.prop

deadYokai, 2023-01-10:
try a recovery.img instead boot.img, it helped me

ghostranger047, 2023-01-12:
> try a recovery.img instead boot.img, it helped me

My device is Treble with A/B partitions.
There is no recovery partition for such devices hence no separate recovery.img.
Recovery is built into boot.img

MLXProjects, 2023-01-15:
the same when inspecting a TWRP image just dumped from my device

ghostranger047, 2023-02-15:
Useing vendor_boot.img and not boot.img fixed the issue.

closed 2023-02-15 by ghostranger047