twrpdtgen/twrpdtgen #150: Generate tree in system.

caiiiok254, 2023-04-04:
Looking through the repository, I came across a TODO file which says that it is planned to generate a tree directly from the system, which would be very useful because my phone has 3 recovery sections:
Recovery (kernel + fstab file)
Recovery_ramdisk (partition that is usually flashed to support custom recovery)
Recovery_vendor (partition with only one vendor folder, in which the drivers and the default.prop file are located)
And in order for the software to work, I will need to combine these sections together manually, repacking the stock recovery, which can lead to errors as a result. While in the working system all these directories are already mounted and, if root is available, are quite accessible. Therefore, if you have any solution or developments in this direction, then I would be happy to help.

SebaUbuntu, 2024-02-11:
No support for Huawei phones

closed 2024-02-11 by SebaUbuntu

ItsCOMMANDer, 2024-12-30:
> No support for Huawei phones

Why if i may ask?

SebaUbuntu, 2024-12-31:
> > No support for Huawei phones
> 
> Why if i may ask?

Too hard to maintain the code for the whole 3 users with an unlocked Huawei device

ItsCOMMANDer, 2025-01-02:
Fair, as someone with a unlocked huawei device, Huawei's a bitch