twrpdtgen/twrpdtgen #152: practically empty boot.img; unable to unpack vendor_boot.img

fukiame, 2023-04-13:
Images dumped by mtkclient
Android 11, seems to be A/B

```
python3 -m twrpdtgen boot.img
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/fukiame/.local/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/fukiame/.local/lib/python3.10/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
  File "/home/fukiame/.local/lib/python3.10/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
  File "/home/fukiame/.local/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 117, in __init__
    self.codename = self.get_first_prop(DEVICE_CODENAME)
  File "/home/fukiame/.local/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 177, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.bootimage.device could not be found in build.prop
```
after manually unpacking the boot.img myself, there are just empty directory and init
trying vendor_boot.img (according to #143 ) gave
```
python3 -m twrpdtgen vendor_boot.img
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/fukiame/.local/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/fukiame/.local/lib/python3.10/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
  File "/home/fukiame/.local/lib/python3.10/site-packages/twrpdtgen/device_tree.py", line 56, in __init__
    self.image_info = self.aik_manager.unpackimg(image)
  File "/home/fukiame/.local/lib/python3.10/site-packages/sebaubuntu_libs/libaik/__init__.py", line 122, in unpackimg
    raise RuntimeError(f"AIK extraction failed, return code {returncode}")
RuntimeError: AIK extraction failed, return code 1
```

Image files: https://u.pcloud.link/publink/show?code=kZFD5KVZ3MEEKqDxYGklNpMs2xtfF0U7slp7

fukiame, 2023-04-13:
Magisk-patching the boot.img still works as expected, though, weird

skysumbra, 2023-04-28:
This tool use unpack from [this](https://github.com/SebaUbuntu/AIK-Linux-mirror)
umm... your boot file not support unpack or injure.
```
Android Image Kitchen - UnpackImg Script
by osm0sis @ xda-developers
 
Supplied image: vendor_boot.img
 
Setting up work folders...
 
Unrecognized format.
```

Maybe you can try[ other tool](https://forum.xda-developers.com/t/tool-android-image-kitchen-unpack-repack-kernel-ramdisk-win-android-linux-mac.2073775/post-36068129) check it again...

SebaUbuntu, 2024-02-11:
You have to use aospdtgen for newer Android versions

closed 2024-02-11 by SebaUbuntu