twrpdtgen/twrpdtgen #156: Many Critical files end up being broken symlinks when compiled.

sounddrill31, 2023-05-21:
- [NA] I am running an official build of TWRP, downloaded from https://twrp.me/Devices/
- [Yes*] I am running the latest version of TWRP
- [Yes] I have read the FAQ (https://twrp.me/FAQ/)
- [Yes*] I have searched for my issue and it does not already exist

**Device codename**: A109
**TWRP version**: NA, 3.7.0-9.0
**Device tree**: https://github.com/sounddrill31/android_device_micromax_a109/
#### WHAT STEPS WILL REPRODUCE THE PROBLEM?
Build TWRP with minimal manifests omni and android 9 branch(also happens in 7.1 branch)

#### WHAT IS THE EXPECTED RESULT?
Recovery should atleast boot.

#### WHAT HAPPENS INSTEAD?
Recovery does not boot. 

#### ADDITIONAL INFORMATION
Upon unpacking the image with carliv, we can see ```!<symlink>ÿþp r o p . d e f a u l t   ``` with NULL symbols. This happens with multiple files like default.prop, charger, bin, etc. 

Vscode web shows that the files are symlinked, and half of them don't lead anywhere. 

I will post logs in a bit. 
`/tmp/recovery.log`: 
`dmesg`: 

Disclaimer: I used venv for python2. The command I used:
```
sudo apt install -y python2
virtualenv --python=python2 ~/.lineage_venv
source ~/.lineage_venv/bin/activate
sudo ln -sf  /home/gitpod/.lineage_venv/bin/python /usr/bin/python
```
[product_copy_files_ignored.txt](https://github.com/TeamWin/android_bootable_recovery/files/11522744/product_copy_files_ignored.txt)
[soong.log](https://github.com/TeamWin/android_bootable_recovery/files/11522746/soong.log)

[android-info.txt](https://github.com/TeamWin/android_bootable_recovery/files/11522745/android-info.txt)

Crossposted from https://github.com/TeamWin/android_bootable_recovery/issues/313 since I see this with other recovery projects as well.

sounddrill31, 2023-06-25:
This issue seems to no longer persist, but my recovery is still not booting. Any help is appreciated.

SebaUbuntu, 2024-02-11:
Once the tree is generated, you're on your own

closed 2024-02-11 by SebaUbuntu