twrpdtgen/twrpdtgen #158: AssertionError: Property ro.product.bootimage.device could not be found in build.prop

Finite-Code, 2023-06-29:
DEVICE := vicky (Motorola Moto G72)
chipset := Helio G99
cmdline prompt := python3 -m twrpdtgen boot.img

The same error has been discussed about two to three times so I dont see any need for pasting it below as it is the exact same!

Finite-Code, 2023-06-29:
Please provide some valid solution i checked the extracted the stock boot.img and in the build.prop the value is present

Finite-Code, 2023-07-02:
I expect some response...

SebaUbuntu, 2023-07-02:
First of all, expect? This is not how open source works, I'm not paid to do this.

Second, didn't you read the big warning in the README? Your device surely has an Android version newer than Oreo, so you should use aospdtgen

If you really want this fixed, you should send at least attach a boot.img

Finite-Code, 2023-07-02:
Dear SebaUbuntu,

Thank you for taking the time to respond to my previous message, and I apologize if my previous statement came across as demanding or entitled. I'm still learning and getting acquainted with the open-source community, so I appreciate your understanding.

Regarding your suggestion to use the aospdtgen tool, I did give it a try, but unfortunately, I encountered the same error as before. That's why, out of curiosity and desperation to find a solution, I sought advice from the EvoX group. I understand now that it would have been better to stick strictly to the instructions in the README file.

I accept responsibility for my oversight in not including the boot.img file in my initial message. I apologize for any inconvenience caused by this omission. If you're still willing to assist me, I will provide the necessary boot.img file for further analysis.

Once again, I'm sorry for any unintended rudeness in my previous message. I truly appreciate any help or guidance you can offer. Please let me know if there is any additional information you need from my end to better understand the issue.

Thank you for your patience and understanding.

And here's the boot.img -
[boot-img-unzip-it-dont-rename.zip](https://github.com/twrpdtgen/twrpdtgen/files/11929844/boot-img-unzip-it-dont-rename.zip)

Finite-Code, 2023-07-08:
Solved By Using vendor_boot.img for compiling instead of  boot.img; 
Any other questions related to the same contact FiniteCode on Telegram!
And thanks to @SebaUbuntu for their tool; And sorry for any misbehavior
You may close this issue if you want!

closed 2023-07-21 by Finite-Code