twrpdtgen/twrpdtgen #161: `twrpdtgen` does not work in Python 3.12

soulspark666, 2023-07-15:
While trying to execute 

```python
 ~\Desktop\dt> python -m twrpdtgen .\boot.img
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\Animesh\AppData\Local\Programs\Python\Python312\Lib\site-packages\twrpdtgen\main.py", line 8, in <module>
    from twrpdtgen.main import main
  File "C:\Users\Animesh\AppData\Local\Programs\Python\Python312\Lib\site-packages\twrpdtgen\main.py", line 11, in <module>
    from twrpdtgen.device_tree import DeviceTree
  File "C:\Users\Animesh\AppData\Local\Programs\Python\Python312\Lib\site-packages\twrpdtgen\device_tree.py", line 12, in <module>
    from sebaubuntu_libs.libandroid.device_info import DeviceArch, DeviceInfo
  File "C:\Users\Animesh\AppData\Local\Programs\Python\Python312\Lib\site-packages\sebaubuntu_libs\libandroid\device_info.py", line 7, in <module>
    from distutils.util import strtobool
ModuleNotFoundError: No module named 'distutils'
```

Upon further [investigation](https://stackoverflow.com/a/76691103), it seems:
[image]

Where it can bee seen that `distutils` is deprecated: https://peps.python.org/pep-0632/

So it seems this will not hold true now:
[image]

Shubthedeveloper, 2023-12-25:
Remove latest python
Reinstall python 3.12 
Try pip install  twrpgen

SebaUbuntu, 2024-07-05:
Fixed in sebaubuntu_libs

closed 2024-07-05 by SebaUbuntu