twrpdtgen/twrpdtgen #162: NotImplementedError: Windows is not supported

Singala929, 2023-07-27:
I got recovery image of HTC Desire 12 and I got this error, Am I doing something wrong?
```
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\-\Downloads\1\twrpdtgen-master\twrpdtgen\__main__.py", line 11, in <module>
    main()
  File "C:\Users\-\Downloads\1\twrpdtgen-master\twrpdtgen\main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\-\Downloads\1\twrpdtgen-master\twrpdtgen\device_tree.py", line 55, in __init__
    self.aik_manager = AIKManager()
                       ^^^^^^^^^^^^
  File "C:\Users\-\AppData\Local\Programs\Python\Python311\Lib\site-packages\sebaubuntu_libs\libaik\__init__.py", line 92, in __init__
    raise NotImplementedError(f"{system()} is not supported")
NotImplementedError: Windows is not supported
```

SebaUbuntu, 2023-07-27:
Yes, Windows is not supported

closed 2023-07-27 by SebaUbuntu

Fristender, 2023-12-21:
> Yes, Windows is not supported
Then why does https://github.com/twrpdtgen/twrpdtgen/wiki/0.-What-you-need say you need a PC with Linux or Windows? Please update it.

SebaUbuntu, 2023-12-21:
> > Yes, Windows is not supported
> > Then why does https://github.com/twrpdtgen/twrpdtgen/wiki/0.-What-you-need say you need a PC with Linux or Windows? Please update it.

Done