twrpdtgen/twrpdtgen #163: AssertionError: Property ro.build.version.security_patch could not be found in build.prop

tame1d, 2023-08-12:
python -m twrpdtgen recovery.img -d

```python
TWRP device tree generator
Version 3.0.0

[__init__.py:104 INFO] __init__: Cloning AIK...
[cmd.py:976 DEBUG] execute: Popen(['git', 'clone', '-v', '--', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/tmp/tmpfnak21c0'], cwd=/home/tame/Desktop/dtgen, universal_newlines=True, shell=None, istream=None)
[base.py:1235 DEBUG] _clone: Cmd(['git', 'clone', '-v', '--', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/tmp/tmpfnak21c0'])'s unused stdout: 
[device_tree.py:58 DEBUG] __init__: Getting device infos...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/tame/.local/lib/python3.11/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/tame/.local/lib/python3.11/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tame/.local/lib/python3.11/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tame/.local/lib/python3.11/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 164, in __init__
    self.build_security_patch = self.get_first_prop(BUILD_SECURITY_PATCH)
                                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tame/.local/lib/python3.11/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 177, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.build.version.security_patch could not be found in build.prop

```
here's the recovery image i used: https://file.io/bGKk7gNiMVba

duarteoficialhp, 2024-02-24:
- [ ] - - [ ]

s2wcn, 2024-04-20:
[image]

s2wcn, 2024-04-20:
```
python3 -m twrpdtgen /root/twrp/boot.img -d
TWRP device tree generator
Version 3.0.0

[__init__.py:108 INFO] __init__: Cloning AIK...
[cmd.py:1253 DEBUG] execute: Popen(['git', 'clone', '-v', '--', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/tmp/tmpfr5zc07w'], cwd=/root/twrp, stdin=None, shell=False, universal_newlines=True)
[base.py:1395 DEBUG] _clone: Cmd(['git', 'clone', '-v', '--', 'https://github.com/SebaUbuntu/AIK-Linux-mirror', '/tmp/tmpfr5zc07w'])'s unused stdout: Cloning into '/tmp/tmpfr5zc07w'...

[device_tree.py:58 DEBUG] __init__: Getting device infos...
Traceback (most recent call last):
  File "/root/miniconda3/envs/py310/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/root/miniconda3/envs/py310/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/root/miniconda3/envs/py310/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/root/miniconda3/envs/py310/lib/python3.10/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
  File "/root/miniconda3/envs/py310/lib/python3.10/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
  File "/root/miniconda3/envs/py310/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 166, in __init__
    self.first_api_level = self.get_first_prop(FIRST_API_LEVEL)
  File "/root/miniconda3/envs/py310/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 185, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.first_api_level could not be found in build.prop
```