twrpdtgen/twrpdtgen #172: PRODUCT_STATIC_BOOT_CONTROL_HAL is obsolete

Zoli06, 2023-09-18:
I'm trying to generate a device tree for my Motorola Moto G32 (devon) but `lunch` command fails:
```
[zoltan@asus-laptop]~/Documents/motorola-root/twrp% lunch omni_devon-eng
In file included from build/make/core/config.mk:313:
In file included from build/make/core/envsetup.mk:312:
In file included from build/make/target/product/telephony_vendor.mk:24:
device/motorola/devon/device.mk:24: error: PRODUCT_STATIC_BOOT_CONTROL_HAL is obsolete. Use shared library module instead. See https://android.googlesource.com/platform/build/+/master/Changes.md#PRODUCT_STATIC_BOOT_CONTROL_HAL.
08:55:31 dumpvars failed with: exit status 1
Device devon not found. Attempting to retrieve device repository from TeamWin Github (http://github.com/TeamWin).
Repository for devon not found in the TeamWin Github repository list.
If this is in error, you may need to manually add it to your .repo/local_manifests/roomservice.xml
In file included from build/make/core/config.mk:313:
In file included from build/make/core/envsetup.mk:312:
In file included from build/make/target/product/telephony_vendor.mk:24:
device/motorola/devon/device.mk:24: error: PRODUCT_STATIC_BOOT_CONTROL_HAL is obsolete. Use shared library module instead. See https://android.googlesource.com/platform/build/+/master/Changes.md#PRODUCT_STATIC_BOOT_CONTROL_HAL.
08:55:32 dumpvars failed with: exit status 1
In file included from build/make/core/config.mk:313:
In file included from build/make/core/envsetup.mk:312:
In file included from build/make/target/product/telephony_vendor.mk:24:
device/motorola/devon/device.mk:24: error: PRODUCT_STATIC_BOOT_CONTROL_HAL is obsolete. Use shared library module instead. See https://android.googlesource.com/platform/build/+/master/Changes.md#PRODUCT_STATIC_BOOT_CONTROL_HAL.
08:55:32 dumpvars failed with: exit status 1

** Don't have a product spec for: 'omni_devon'
** Do you have the right repo manifest?
```

These are what I found:
https://android.googlesource.com/platform/build/+/a2d19cfd059933cd287139932e7bec52cf540c65/Changes.md#build-system-changes-for-android_mk-writers
https://github.com/minimal-manifest-twrp/platform_manifest_twrp_aosp/issues/32#issuecomment-1684904058

Any help is appreciated.

HighlanderUkUk, 2024-01-10:
look in the device.mk file and edit out the PRODUCT_STATIC_BOOT_CONTROL_HAL lines

SebaUbuntu, 2024-02-11:
Assuming this device is new enough to hit that issue, use aospdtgen

closed 2024-02-11 by SebaUbuntu