twrpdtgen/twrpdtgen #173: AIK extraction failed, return code 1

sali9043, 2023-09-20:
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/twrpdtgen/device_tree.py", line 56, in __init__
    self.image_info = self.aik_manager.unpackimg(image)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/sebaubuntu_libs/libaik/__init__.py", line 122, in unpackimg
    raise RuntimeError(f"AIK extraction failed, return code {returncode}")
RuntimeError: AIK extraction failed, return code 1

SebaUbuntu, 2024-02-07:
Please report it to the AIK developer

closed 2024-02-07 by SebaUbuntu

AllanFuego, 2025-03-22:
> TWRP device tree generator Version 3.0.0
> 
> [INFO] Cloning AIK... Traceback (most recent call last): File "", line 198, in _run_module_as_main File "", line 88, in _run_code File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/twrpdtgen/**main**.py", line 11, in main() File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/twrpdtgen/main.py", line 37, in main device_tree = DeviceTree(image=args.image) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/twrpdtgen/device_tree.py", line 56, in **init** self.image_info = self.aik_manager.unpackimg(image) ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ File "/data/data/com.termux/files/usr/lib/python3.11/site-packages/sebaubuntu_libs/libaik/**init**.py", line 122, in unpackimg raise RuntimeError(f"AIK extraction failed, return code {returncode}") RuntimeError: AIK extraction failed, return code 1

Follow these steps(i have compiled device tree for redmi note 8 from termux)

1) pip3 install twrpdtgen
2) pip3 show twrpdtgen
If the installation is successfull you will see something like these : 
Name: twrpdtgen
Version: 3.0.0
Summary: A Python library/script to automatically generate TWRP-compatible device tree from a boot/recovery image
Home-page: https://github.com/twrpdtgen/twrpdtgen
Author: Sebastiano Barezzi
Author-email: barezzisebastiano@gmail.com
License: Apache-2.0
Location: /data/data/com.termux/files/usr/lib/python3.12/site-packages
Requires: GitPython, Jinja2, sebaubuntu-libs
Required-by:

3) Grab the Location , in termux its in - /data/data/com.termux/files/usr/lib/python3.12/site-packages
4) Go to the location and do ls and see if the twrpdtgen folder is availablw or not.
5) Once you verified its there go to the twrpdtgen folder and look if __main__.py is present or not.
6) If you find the python file inside the folder you are goot to go👍
7) Link the path to termux bin directory and give it executeable permission so that you can directly use twrpdtgen from anywhere.
"
ln -s /data/data/com.termux/files/usr/lib/python3.12/site-packages/twrpdtgen/__main__.py /data/data/com.termux/files/usr/bin/twrpdtgen
chmod +x /data/data/com.termux/files/usr/bin/twrpdtgen
"
8) Run "twrpdtgen --help"
9) If everything is done perfectly you will see twrpdtgen working perfectly.
10) Install cpio pkg in termux using "pkg install cpio"
11) Now you can build your device tree using "twrpdtgen /path/to/recovery.img"

Hope it helps 👍