twrpdtgen/twrpdtgen #175: AssertionError: Property ro.product.first_api_level could not be found in build.prop

mertoalex, 2023-10-01:
I run `python3 -m twrpdtgen ./recovery.img` command (I renamed `recovery.bin` to `recovery.img` from mtkclient extraction)
and It's gives this error:
```
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/mertoalex/.local/lib/python3.11/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/mertoalex/.local/lib/python3.11/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mertoalex/.local/lib/python3.11/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mertoalex/.local/lib/python3.11/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 163, in __init__
    self.first_api_level = self.get_first_prop(FIRST_API_LEVEL)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/mertoalex/.local/lib/python3.11/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 179, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.first_api_level could not be found in build.prop
```

mtkclient gui output when I plug in my phone:
```
Preloader - 	CPU:			MT6755/MT6750/M/T/S(Helio P10/P15/P18)
Preloader - 	HW version:		0x0
Preloader - 	WDT:			0x10007000
Preloader - 	Uart:			0x11002000
Preloader - 	Brom payload addr:	0x100a00
Preloader - 	DA payload addr:	0x201000
Preloader - 	CQ_DMA addr:		0x10212c00
Preloader - 	Var1:			0xa
Preloader - Disabling Watchdog...
Preloader - HW code:			0x326
Preloader - Target config:		0x0
Preloader - 	SBC enabled:		False
Preloader - 	SLA enabled:		False
Preloader - 	DAA enabled:		False
Preloader - 	SWJTAG enabled:		False
Preloader - 	EPP_PARAM at 0x600 after EMMC_BOOT/SDMMC_BOOTFalse
Preloader - 	Root cert required:	False
Preloader - 	Mem read auth:		False
Preloader - 	Mem write auth:		False
Preloader - 	Cmd 0xC8 blocked:	False
Preloader - Get Target info
Preloader - BROM mode detected.
Preloader - 	HW subcode:		0x8a00
Preloader - 	HW Ver:			0xcb00
Preloader - 	SW Ver:			0x1
Preloader - ME_ID:			9BC52711D18EC2FC390C359A984B94CD
```

renamed .bin file (zipped): [recovery.img.zip](https://github.com/twrpdtgen/twrpdtgen/files/12777974/recovery.img.zip)

mertoalex, 2023-10-01:
btw I tried to install a port of twrp somehow to my phone, that twrp port is broken

mertoalex, 2023-10-01:
accident close

mertoalex, 2023-10-01:
I edited default.prop file and added this line and now It's works ?
```
ro.product.first_api_level=23
```

closed 2023-10-01 by mertoalex

yunzimo, 2024-04-06:
> 我编辑了default.prop文件并添加了这一行，现在它有效吗？
> 
> ```
> ro.product.first_api_level=23
> ```

Have you solved this problem?

mertoalex, 2024-04-07:
> > 我编辑了default.prop文件并添加了这一行，现在它有效吗？
> > ```
> > ro.product.first_api_level=23
> > ```
> 
> Have you solved this problem?

I don't remember anything about this, sorry.

mertoalex, 2025-03-19:
i left my twrp porting adventure at that time because i didn't had enough storage for android, now i have enough storage so i started from scratch to port cyanogenmod+twrp  and i had this error again and  searched now and found out that `ro.product.first_api_level` is just value that shows device's first stock-rom android api level ([src](https://source.android.com/docs/compatibility/cts/setup))
like, my mom's phone is xiaomi mi 8 pro and that phone first shipped with android 8 so `ro.product.first_api_level` is `27` on her phone ([list of api levels based on android version](https://apilevels.com/))

so based on that information, i should add
> ```
> ro.product.first_api_level=23
> ```
to my boot/recovery image's prop file (generally ramdisk/default.prop) and it should fix the error, probably can applied to all boot/recovery images

mertoalex, 2025-03-19:
and fortunately it worked!

[image]

so i can answer @yunzimo 's question
> Have you solved this problem?
yes, i did and explained here now.