twrpdtgen/twrpdtgen #191: UnicodeDecodeError: invalid start byte

4accccc, 2024-02-06:
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "/usr/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/4a/.local/lib/python3.10/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/4a/.local/lib/python3.10/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
  File "/home/4a/.local/lib/python3.10/site-packages/twrpdtgen/device_tree.py", line 64, in __init__
    self.build_prop.import_props(build_prop)
  File "/home/4a/.local/lib/python3.10/site-packages/sebaubuntu_libs/libandroid/props/__init__.py", line 40, in import_props
    text = file.read_text(encoding="utf-8")
  File "/usr/lib/python3.10/pathlib.py", line 1135, in read_text
    return f.read()
  File "/usr/lib/python3.10/codecs.py", line 322, in decode
    (result, consumed) = self._buffer_decode(data, self.errors, final)
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xb0 in position 4205: invalid start byte

this is the recovery image which was dumped by me using mtkclient.
[recovery.zip](https://github.com/twrpdtgen/twrpdtgen/files/14186219/recovery.zip)

someone plz help me!

duarteoficialhp, 2024-02-29:
.github/CONTRIBUTING.md

duarteoficialhp, 2024-04-04:
- ~~~~~~~~@duarteoficialhp # @mauronofrio @tiann ~~~~