twrpdtgen/twrpdtgen #192: error when lanucing

dzwdev, 2024-02-07:
i get this error:
`TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/pc/twrgen1/.venv/lib/python3.11/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/pc/twrgen1/.venv/lib/python3.11/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pc/twrgen1/.venv/lib/python3.11/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pc/twrgen1/.venv/lib/python3.11/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 151, in __init__
    self.bootloader_board_name = self.get_first_prop(BOOTLOADER_BOARD_NAME)
                                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pc/twrgen1/.venv/lib/python3.11/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 179, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.board could not be found in build.prop
`

SebaUbuntu, 2024-02-11:
Blame your OEM for doing weird stuff (OnePlus? Oppo?)

closed 2024-02-11 by SebaUbuntu

dzwdev, 2024-02-11:
Huawei

SebaUbuntu, 2024-02-11:
Yeah, not supported unfortunately