twrpdtgen/twrpdtgen #193: When running the program, returns error about nothing working until I add Git to path.

AblertARock, 2024-02-10:
The error code:
`
Traceback (most recent call last):
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\git\__init__.py", line 140, in <module>
    refresh()
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\git\__init__.py", line 127, in refresh
    if not Git.refresh(path=path):
           ^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\git\cmd.py", line 456, in refresh
    raise ImportError(err)
ImportError: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\twrpdtgen\__main__.py", line 8, in <module>
    from twrpdtgen.main import main
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\twrpdtgen\main.py", line 11, in <module>
    from twrpdtgen.device_tree import DeviceTree
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\twrpdtgen\device_tree.py", line 8, in <module>
    from git import Repo
  File "C:\Users\craig_3v04aan\AppData\Local\Programs\Python\Python312\Lib\site-packages\git\__init__.py", line 142, in <module>
    raise ImportError("Failed to initialize: {0}".format(_exc)) from _exc
ImportError: Failed to initialize: Bad git executable.
The git executable must be specified in one of the following ways:
    - be included in your $PATH
    - be set via $GIT_PYTHON_GIT_EXECUTABLE
    - explicitly set via git.refresh()

All git commands will error until this is rectified.

This initial warning can be silenced or aggravated in the future by setting the
$GIT_PYTHON_REFRESH environment variable. Use one of the following values:
    - quiet|q|silence|s|none|n|0: for no warning or exception
    - warn|w|warning|1: for a printed warning
    - error|e|raise|r|2: for a raised exception

Example:
    export GIT_PYTHON_REFRESH=quiet
`
Anything I can do to rectify this?

SebaUbuntu, 2024-02-11:
`The git executable must be specified in one of the following ways`

What do you think?

closed 2024-02-11 by SebaUbuntu

AblertARock, 2024-02-11:
I had added Git to the path. Problem was I needed to add the bin directory as well. However, I ended up with the fact that it doesn't work on Windows, so...