twrpdtgen/twrpdtgen #209: Error when building twrp from source

weiliang011, 2024-06-14:
Hi, i got an error when building twrp, any solution? Here is the error my device is lenovo tb-x606fa with this: [https://github.com/minimal-manifest-twrp/platform_manifest_twrp_aosp/tree/twrp-12.1](url) manifest of twrp because my gsi rom is android 14 and i think is fbe, so here is the error 
`w@w:~/TWRP$ mka recoveryimage
16:05:29 ************************************************************
16:05:29 You are building on a machine with 8.01GB of RAM
16:05:29 
16:05:29 The minimum required amount of free memory is around 16GB,
16:05:29 and even with that, some configurations may not work.
16:05:29 
16:05:29 If you run into segfaults or other errors, try reducing your
16:05:29 -j value.
16:05:29 ************************************************************
16:05:30 Build sandboxing disabled due to nsjail error.
build/make/core/soong_config.mk:194: warning: BOARD_PLAT_PUBLIC_SEPOLICY_DIR has
 been deprecated. Use SYSTEM_EXT_PUBLIC_SEPOLICY_DIRS instead.
build/make/core/soong_config.mk:195: warning: BOARD_PLAT_PRIVATE_SEPOLICY_DIR ha
s been deprecated. Use SYSTEM_EXT_PRIVATE_SEPOLICY_DIRS instead.
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=16.1.0
TARGET_PRODUCT=twrp_X606FA
TARGET_BUILD_VARIANT=eng
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-6.8.0-35-generic-x86_64-Ubuntu-24.04-LTS
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=SP2A.220405.004
OUT_DIR=out
============================================
[ 98% 238/241] test android/soong/cc
[100% 241/241] out/soong/.bootstrap/bin/soong_build out/soong/build.ninja
FAILED: out/soong/build.ninja
cd "$(dirname "out/soong/.bootstrap/bin/soong_build")" && BUILDER="$PWD/$(basena
me "out/soong/.bootstrap/bin/soong_build")" && cd / && env -i "$BUILDER"     --t
op "$TOP"     --out "out/soong"     -n "out"     -d "out/soong/build.ninja.d"   
  -t -l out/.module_paths/Android.bp.list -globFile out/soong/.bootstrap/build-g
lobs.ninja -o out/soong/build.ninja --available_env out/soong/soong.environment.
available --used_env out/soong/soong.environment.used Android.bp
internal error: panic in GenerateBuildActions for module "framework_compatibilit
y_matrix.6.xml" variant "android_common"
runtime error: invalid memory address or nil pointer dereference
goroutine 4606215 [running]:
github.com/google/blueprint.newPanicErrorf(0xd31220, 0x15888f0, 0xc0e4946c60, 0x
5f, 0x0, 0x0, 0x0, 0xc0e495dc00, 0x31)
	/home/w/TWRP/build/blueprint/context.go:4142 +0x7b
github.com/google/blueprint.(*Context).generateModuleBuildActions.func2.1.1(0xc0
26ff9a40, 0xc0e494d5f0)
	/home/w/TWRP/build/blueprint/context.go:2831 +0x3a5
panic(0xd31220, 0x15888f0)
	prebuilts/go/linux-x86/src/runtime/panic.go:969 +0x1b9
github.com/google/blueprint.(*moduleInfo).Name(...)
	/home/w/TWRP/build/blueprint/context.go:300
github.com/google/blueprint.(*baseModuleContext).OtherModuleName(0xc0e494d5f0, 0
x0, 0x0, 0x0, 0x0)
	/home/w/TWRP/build/blueprint/module_ctx.go:489 +0x6d
android/soong/android.(*baseModuleContext).OtherModuleName(0xc0e496a490, 0x0, 0x
0, 0x0, 0x0)
	/home/w/TWRP/build/soong/android/module.go:2019 +0x4a
android/soong/vintf-compatibility-matrix.(*vintfCompatibilityMatrixRule).getSche
ma(0xc0c38b6e00, 0xff42a0, 0xc0e496a480, 0x31, 0xaaaaaaaaaaaaaaaa, 0x38)
[... 129 lines cut ...]
	/home/w/TWRP/hardware/interfaces/compatibility_matrices/build/vintf_comp
atibility_matrix.go:121 +0x2a2
android/soong/vintf-compatibility-matrix.(*vintfCompatibilityMatrixRule).Generat
eAndroidBuildActions(0xc0c2476a00, 0xff42a0, 0xc0e496b440)
	/home/w/TWRP/hardware/interfaces/compatibility_matrices/build/vintf_comp
atibility_matrix.go:140 +0x8d
android/soong/android.(*ModuleBase).GenerateBuildActions(0xc0c2476a00, 0xfe3b60,
 0xc0e4970820)
	/home/w/TWRP/build/soong/android/module.go:1850 +0xc5f
github.com/google/blueprint.(*Context).generateModuleBuildActions.func2.1(0xc03c
c9ea80, 0xc0e4970820)
	/home/w/TWRP/build/blueprint/context.go:2835 +0x85
github.com/google/blueprint.(*Context).generateModuleBuildActions.func2(0xc03cc9
ea80, 0xc0020f41e0, 0x0)
	/home/w/TWRP/build/blueprint/context.go:2836 +0x345
github.com/google/blueprint.parallelVisit.func1.1(0xc0c8c5bf40, 0xc03cc9ea80, 0x
c0020f41e0, 0xc0020f4180, 0xc0020f4120)
	/home/w/TWRP/build/blueprint/context.go:1973 +0x3a
created by github.com/google/blueprint.parallelVisit.func1
	/home/w/TWRP/build/blueprint/context.go:1972 +0x12e

16:09:41 soong bootstrap failed with: exit status 1

#### failed to build some targets (04:13 (mm:ss)) ####
`

Wishmasterflo, 2024-06-26:
Hi!
I have that Lenovo Tablet as well and found a [thread](https://xdaforums.com/t/recovery-tb-x606f-tb-x606fa-unofficial-twrp-3-7-x-for-lenovo-tab-m10-fhd-plus.4222887) on XDA with a working TWRP recovery for it.
I adjusted that recovery tree for Orangefox and could successfully build a working Orangefox recovery for it. 

For FBE to work there is a patch which needs to be applied to the synced recovery before building. 
See [here](https://github.com/Wishmasterflo/recovery_device_lenovo_X606F/blob/android-12.1-fbe/patch/system/vold/0001-Remove-a12-kmblob-magic.patch)

Here is my Working Orangefox recovery for it. I have the X606F but this can easily be adjusted to the X606FA as well.

weiliang011, 2024-06-26:
> Hi! I have that Lenovo Tablet as well and found a [thread](https://xdaforums.com/t/recovery-tb-x606f-tb-x606fa-unofficial-twrp-3-7-x-for-lenovo-tab-m10-fhd-plus.4222887) on XDA with a working TWRP [...]

How to apply the patch?  How I adjust to the x606fa?

Wishmasterflo, 2024-06-27:
[This](https://github.com/Yahoo-Mike/recovery_device_lenovo_X606FA/)  is the TWRP device tree which is by default for the X606FA and it has both Kernel Image files for F and FA Version in the prebuilt folder. 
So the Kernel name to use is just set in the Boardconfig.mk file. 

After have synced TWRP and the device tree you go into the folder /twrp/system/vold
and do:
**git apply /home/flo/twrp/device/lenovo/X606FA/patch/system/vold/0001-Remove-a12-kmblob-magic.patch**

(just change the home folder name and username to the local location where you had synced the device tree and twrp!)

Also I had to add TW_INCLUDE_RESETPROP := true  to Boardconfig.mk when I build Orangefox recovery.
Otherwise it would not boot up.
Had also added logging so that a logcat etc can get fetched to Boardconfig.mk 

TWRP_INCLUDE_LOGCAT := true
TARGET_USES_LOGD := true

SebaUbuntu, 2024-07-05:
From the error, it looks like you don't have enough RAM