twrpdtgen/twrpdtgen #217: Property {props[0]} could not be found in build.prop

Athwale, 2024-09-18:
Hi,
I am attempting to use this tool with this ROM https://xdaforums.com/t/soyes-s23-pro-mini-chinese-phone-from-aliexpress.4681069/#post-89718376
And I am running into problems:

$ python3 -m twrpdtgen ../ROM/FW/recovery.img 
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/tt/.local/lib/python3.12/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
  File "/home/tt/.local/lib/python3.12/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tt/.local/lib/python3.12/site-packages/twrpdtgen/device_tree.py", line 66, in __init__
    self.device_info = DeviceInfo(self.build_prop)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tt/.local/lib/python3.12/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 166, in __init__
    self.first_api_level = self.get_first_prop(FIRST_API_LEVEL)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tt/.local/lib/python3.12/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 185, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')

Any idea what is wrong?
I have noticed that the build.prop is actually in the system.img in this firmware. I have no idea why.
It is Android 8.1.

Athwale, 2024-09-18:
I seem to have solved the problem by injecting the API level like this:
```
if 'prop.default' in str(build_prop):
	with open(build_prop, 'a') as f:
		f.write('ro.product.first_api_level=27')
```
On line 64 in device_tree.py
27 for Android 8.1

schrmh, 2025-02-05:
Hey, line 64 is a continue statement and replacing that or putting that below did not work for me in getting something else than your error with the ROM you linked to. Could you send a diff or specify it a bit different?

Athwale, 2025-02-05:
Hi, here is the whole block of code which should make it clear.

```
54         # Extract the image
 55         self.aik_manager = AIKManager()
 56         self.image_info = self.aik_manager.unpackimg(image)
 57 
 58         LOGD("Getting device infos...")
 59         self.build_prop = BuildProp()
 60         for build_prop in [self.image_info.ramdisk / location for location in BUILDPROP_LOCATIONS]:
 61             if not build_prop.is_file():
 62                 continue
 63 
 64             if 'prop.default' in str(build_prop):
 65                 with open(build_prop, 'a') as f:
 66                     f.write('ro.product.first_api_level=27')
 67             self.build_prop.import_props(build_prop)
 68 
 69         self.device_info = DeviceInfo(self.build_prop)
 70 
 71         # Generate fstab
```

schrmh, 2025-02-05:
Thanks for the quickly reply.
So you basically got rid of `assert self.image_info.ramdisk, "Ramdisk not found"` from line 58 as well.

Well... With the boot.img from the ROM you linked I get this now:
```
TWRP device tree generator
Version 3.0.0

[INFO] Cloning AIK...
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/duda/.local/lib/python3.13/site-packages/twrpdtgen/__main__.py", line 11, in <module>
    main()
    ~~~~^^
  File "/home/duda/.local/lib/python3.13/site-packages/twrpdtgen/main.py", line 37, in main
    device_tree = DeviceTree(image=args.image)
  File "/home/duda/.local/lib/python3.13/site-packages/twrpdtgen/device_tree.py", line 69, in __init__
    self.device_info = DeviceInfo(self.build_prop)
                       ~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/duda/.local/lib/python3.13/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 120, in __init__
    self.codename = self.get_first_prop(DEVICE_CODENAME)
                    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/home/duda/.local/lib/python3.13/site-packages/sebaubuntu_libs/libandroid/device_info.py", line 185, in get_first_prop
    raise AssertionError(f'Property {props[0]} could not be found in build.prop')
AssertionError: Property ro.product.system.device could not be found in build.prop
```

Was this the same for you (aka. additional lines have been needed anyways)?

Athwale, 2025-02-05:
No I did not delete any lines. Just added the few lines there. And did do this with recovery.img

schrmh, 2025-02-05:
Yeah, sorry. It works with recovery.img... When I tried it before with the one for my device I also used recovery.img but I was following some guide that used some kind of combined boot.img (instead of recovery.img) and probably was just not paying attention enough to your first post in this issue.

schrmh, 2025-02-05:
Interestingly, I do actually still get the error with a recovery I [extracted](https://github.com/schrmh/AndroidDevices/blob/master/MT8127/Trekstor/SurfTab%C2%AE%20breeze%2010.1%20quad%20plus/root/MT8127_Android_scatter_trekstor_surftab_quad_101_plus.txt):
[ST10408-11.zip](https://github.com/user-attachments/files/18674229/ST10408-11.zip) (ROM_9 is boot; ROM_10 recovery)

Athwale, 2025-02-05:
Both recovery and boot can be unpacked and packed with https://github.com/LowTension/carliv_image_kitchen/tree/linux64

It looks like your files do not have a prop.default file at all.
I am not sure why that is.
If you are just trying to get twrp, you can try this and see if it works.
https://www.hovatek.com/twrpbuilder/

Athwale, 2025-02-05:
By the way that is another thing that would be nice to have. If a file is not found, show an error and not a traceback,

schrmh, 2025-02-05:
Thanks for the hints. The Carliv thing I used five years ago (and I don't recall how much success I had with that but I definitely never ended with a working twrp tho I also recall that there was some weird fstab layout that I might have gotten wrong somehow) but I just wanted to go with twrpdtgen cause it was used in a newer tutorial. But well, I can try it again. The online builder appears to only support Android 7+.

Athwale, 2025-02-05:
They say that 5.1 is on hold, but I would say it worth trying just in case it is fixed.

The prop.default file should be somewhere, maybe it is in a different image or the structure of you rom is in some way different.
You might get lucky if you find it and just copy it into the recovery image a repack it and then use the generator.

Athwale, 2025-02-05:
System.img is usually an ext4 image so you should be able to just mount it for example.
sudo mount -t ext4 -o loop,rw,noatime system.img mounted/

Athwale, 2025-02-05:
@schrmh I got it to work with your recovery.img
In your case it is using default.prop and you are missing another string there.
I put in one from Android 8.1 because I do not know what it should be for your device.
Here is the modified code:

```
		# Extract the image
		self.aik_manager = AIKManager()
		self.image_info = self.aik_manager.unpackimg(image)

		LOGD("Getting device infos...")
		self.build_prop = BuildProp()
		for build_prop in [self.image_info.ramdisk / location for location in BUILDPROP_LOCATIONS]:
			print(build_prop)
			if not build_prop.is_file():
				print('skip')
				continue

			if 'default.prop' in str(build_prop):
				with open(build_prop, 'a') as f:
					print('open')
					f.write('ro.product.first_api_level=22\nro.build.version.security_patch=2023-10-05')
			print('importing: ', build_prop)
			self.build_prop.import_props(build_prop)

		self.device_info = DeviceInfo(self.build_prop)

		# Generate fstab
```

You can see from the modified output where it is looking for the other files to use:
```
/tmp/tmpw2fxecfl/ramdisk/default.prop
open
importing:  /tmp/tmpw2fxecfl/ramdisk/default.prop
/tmp/tmpw2fxecfl/ramdisk/prop.default
skip
/tmp/tmpw2fxecfl/ramdisk/system/build.prop
skip
/tmp/tmpw2fxecfl/ramdisk/vendor/build.prop
skip
/tmp/tmpw2fxecfl/ramdisk/system/etc/build.prop
skip
/tmp/tmpw2fxecfl/ramdisk/vendor/etc/build.prop
skip
```
Maybe he security patch string could be in one of those files in the respective vendor or system img files.

Athwale, 2025-02-05:
Your boot default.prop looks like this:
```
#
# ADDITIONAL_DEFAULT_PROPERTIES
#
persist.sys.usb.config=mtp
persist.service.acm.enable=0
ro.secure=1
ro.allow.mock.location=0
ro.debuggable=0
ro.zygote=zygote32
camera.disable_zsl_mode=1
ro.mount.fs=EXT4
dalvik.vm.dex2oat-Xms=64m
dalvik.vm.dex2oat-Xmx=512m
dalvik.vm.image-dex2oat-Xms=64m
dalvik.vm.image-dex2oat-Xmx=64m
ro.dalvik.vm.native.bridge=0
```

And recovery default.prop like this:
```
#
# ADDITIONAL_BUILD_PROPERTIES
#
ro.com.android.dateformat=MM-dd-yyyy
ro.config.ringtone=Ring_Synth_04.ogg
ro.config.notification_sound=pixiedust.ogg
ro.carrier=unknown
ro.config.alarm_alert=Alarm_Classic.ogg
drm.service.enabled=true
ro.mtk_in_house_tee_support=true
ro.mediatek.platform=MT8127
ro.adb.secure=1
persist.radio.combo.chipid=-1
mediatek.wlan.chip=mediatek.wlan.module.postfix=_
ro.btstack=blueangel
ro.frp.pst=emmc@frp
ro.mediatek.version.branch=L1
ro.mediatek.version.sdk=4
ro.mtk_audio_profiles=1
ro.mtk_audenh_support=1
ro.mtk_besloudness_support=1
ro.mtk_wapi_support=1
ro.mtk_bt_support=1
ro.mtk_wappush_support=1
ro.mtk_voice_ui_support=1
ro.mtk_voice_unlock_support=1
ro.mtk_voice_contact_support=1
ro.mtk_matv_analog_support=1
ro.mtk_wlan_support=1
ro.mtk_ipo_support=1
ro.mtk_gps_support=1
ro.mtk_search_db_support=1
ro.mtk_dialer_search_support=1
ro.mtk_dhcpv6c_wifi=1
ro.have_aacencode_feature=1
ro.mtk_fd_support=1
ro.mtk_cta_drm_support=1
ro.mtk_widevine_drm_l3_support=1
ro.mtk_flv_playback_support=1
ro.mtk_hdmi_support=1
ro.mtk_mtkps_playback_support=1
ro.mtk_send_rr_support=1
ro.mtk_emmc_support=1
ro.mtk_tetheringipv6_support=1
ro.mtk_phone_number_geo=1
ro.mtk_shared_sdcard=1
ro.mtk_flight_mode_power_off_md=1
ro.mtk_pq_support=2
ro.mtk_wfd_support=1
ro.mtk_wfd_sink_support=1
ro.mtk_wfd_sink_uibc_support=1
ro.mtk_wifi_mcc_support=1
ro.mtk_system_update_support=1
ro.mtk_sim_hot_swap=1
ro.mtk_thumbnail_play_support=1
ro.mtk_cmcc_ft_precheck_support=1
ro.mtk_wfd_hdcp_tx_support=1
ro.mtk_wfd_hdcp_rx_support=1
ro.mtk_world_phone_policy=0
ro.mtk_perfservice_support=1
ro.mtk_audio_change_support=1
ro.mtk_hdmi_hdcp_support=1
ro.mtk_internal_hdmi_support=1
ro.mtk_owner_sdcard_support=1
ro.mtk_owner_sim_support=1
ro.mtk_cta_set=1
ro.mtk_multi_patition=1
ro.mtk_drm_key_mng_support=1
ro.mtk_mobile_management=1
ro.mtk_antibricking_level=2
ro.mtk_cam_mfb_support=1
ro.mtk_cam_mav_support=1
ro.mtk_subtitle_support=1
ro.mtk_live_photo_support=1
ro.mtk_motion_track_support=1
ro.mtk_passpoint_r1_support=1
ro.mtk_bg_power_saving_support=1
ro.mtk_bg_power_saving_ui=1
ro.have_aee_feature=1
ro.sim_me_lock_mode=0
ro.mtk_dual_mic_support=0
ro.mtk_is_tablet=1
ro.mtk_voice_extension_support=1
wfd.dummy.enable=1
ro.mediatek.project.path=device/htt/htt8127_tb_l1
ro.mtk_sec_video_path_support=1
persist.mtk.wcn.combo.chipid=-1
service.wcn.driver.ready=no
ro.com.android.mobiledata=false
persist.radio.mobile.data=0,0
persist.meta.dumpdata=0
ro.sf.hwrotation=0
dalvik.vm.heapstartsize=5m
dalvik.vm.heapgrowthlimit=128m
dalvik.vm.heapsize=256m
dalvik.vm.heaptargetutilization=0.75
dalvik.vm.heapminfree=512k
dalvik.vm.heapmaxfree=2m
persist.sys.multiwindow=1
ro.mediatek.chip_ver=S01
ro.mediatek.version.release=ALPS.L1.MP8.V2_HTT8127.TB.L1
ro.chivin.earpiece_disable=1
ro.mtk_default_ime=com.android.inputmethod.latin.LatinIME
ro.setupwizard.mode=OPTIONAL
ro.com.google.gmsversion=5.1_r3
persist.sys.dalvik.vm.lib.2=libart.so
dalvik.vm.isa.arm.features=div
net.bt.name=Android
dalvik.vm.stack-trace-file=/data/anr/traces.txt
```

So it is definitely not in these.

schrmh, 2025-02-06:
Hmm... Could it be that there is just no ro.build.version.security_patch for this device? It appears that string only exist since Lollipop after all, at least according to some search results I found.
It's nowhere in getprop or in logs I collected either and it doesn't show up in settings and I grep'ed recursively over / (until adb terminated)
adb shell getprop | grep build:
```
[ro.build.OTA.version]: [V1.1.8]
[ro.build.characteristics]: [tablet]
[ro.build.date.utc]: [1457083825]
[ro.build.date]: [Fri Mar  4 17:30:25 CST 2016]
[ro.build.description]: [full_htt8127_tb_l1-user 5.1 LMY47I 1457083674 test-keys]
[ro.build.display.id]: [ST10408-11_20160304_V1.1.8]
[ro.build.fingerprint]: [TrekStor/ST10408-11/ST10408-11:5.1/LMY47I/1457083674:user/test-keys]
[ro.build.flavor]: [full_htt8127_tb_l1-user]
[ro.build.host]: [chivin123-All-Series]
[ro.build.id]: [LMY47I]
[ro.build.product]: [htt8127_tb_l1]
[ro.build.tags]: [test-keys]
[ro.build.type]: [user]
[ro.build.user]: [chivin123]
[ro.build.version.all_codenames]: [REL]
[ro.build.version.codename]: [REL]
[ro.build.version.incremental]: [1457083674]
[ro.build.version.release]: [5.1]
[ro.build.version.sdk]: [22]
```

Or well, complete getprop and some logs that I collected in the past:
[ST10408-11_logs.zip](https://github.com/user-attachments/files/18684393/ST10408-11_logs.zip)
[getprop.txt](https://github.com/user-attachments/files/18684381/getprop.txt)

The default.prop you posted does exist 1:1 at /default.prop; [build.prop](https://github.com/user-attachments/files/18686475/build.prop.txt) was at /system/build.prop (and also in a backup of the system partition).

Again, thanks a lot so far. Will try building TWRP later today.

Athwale, 2025-02-06:
I do not know.
I also do not know how important that line is for building TWRP since it is not full Android.
You might just get away with making some date up so it does not complain.

Athwale, 2025-02-06:
Drop a comment if you are successful with the build. I am just curious since it is even older version.

schrmh, 2025-02-07:
(Okay, one interesting bit: I still had some files from the previous attempt with Carliv laying around and noticed that it created an empty recovery.fstab while the twrpdtgen with your modification produced a file with content.)

I had some issues but eventually I was able to build a recovery.img:
https://xdaforums.com/t/guide-zero-to-hero-what-to-do-when-your-phone-has-nothing-no-twrp-no-custom-roms.4627809/post-89944134
Maybe some issues could have been avoided if I just used Ubuntu 14.04 or 12. In fact, it took a while before I really found the first solution on SO (kinda ignored it in search results cause I thought it probably wouldn't be helpful, heh) so I did try the [Dockerfile](https://android.googlesource.com/platform/build/+/main/tools/docker/Dockerfile) mentioned at https://source.android.com/docs/setup/start/older-versions so that I could just try building with an older ready-to-go Ubuntu but their Dockerfile is just broken and I did not manage to get it working without complaining about invalid user or invalid group.

So yeah, I actually flashed this to the boot partition cause I have a bit of key issues (need to open the device again and put them back so that they will work for a few months again) **but it does boot into TWRP and touch is working!** 🥳 

Oh btw.: the reason I used ST10408_11 instead of ST10408-11 is that _lunch_ apparently can't handle a - in a device name...

Again, thanks a lot. Your comments has been quite the motivation to keep on going with this.
I might put out each step I took a bit more clearly in a few days at https://github.com/schrmh/AndroidDevices/tree/master/MT8127/Trekstor/SurfTab%C2%AE%20breeze%2010.1%20quad%20plus.

Next thing I want to get onto that thing is Android 6 (I think the modern variant of Shizuku works there and in general lots of tools require it...). Ultimatively a postmarketOS port would be nice but I guess that's gonna be hard since I couldn't even find anything close to this device that has kernel sources available — couldn't even find a TWRP I could boot by luck and I think the only other device with this SoC that really shipped with Android 5.1 ootb was a Amazon tablet...

Athwale, 2025-02-07:
Good job. I am glad you managed to make it compile.
Good luck.