twrpdtgen/twrpdtgen #221: Moto G Play 2024 - Recovery.img Not Being Made

jasonmerc, 2024-11-19:
When compiling for a Moto G Play 2024 (fogona), mka recoveryimage doesn't put a recovery.img in out/target/product/fogona.  Not sure why.  Any help would be appreciated.

TWRP source tree can be found here: https://github.com/minimal-manifest-twrp/platform_manifest_twrp_aosp/tree/twrp-12.1

Device tree can be found on my profile: https://github.com/jasonmerc/twrp_device_motorola_fogona

CLI output:

```
me@ligma:~/twrp-121$ mka recoveryimage
build/make/core/soong_config.mk:194: warning: BOARD_PLAT_PUBLIC_SEPOLICY_DIR has been deprecated. Use SYSTEM_EXT_PUBLIC_SEPOLICY_DIRS instead.
build/make/core/soong_config.mk:195: warning: BOARD_PLAT_PRIVATE_SEPOLICY_DIR has been deprecated. Use SYSTEM_EXT_PRIVATE_SEPOLICY_DIRS instead.
============================================
PLATFORM_VERSION_CODENAME=REL
PLATFORM_VERSION=16.1.0
TARGET_PRODUCT=twrp_fogona
TARGET_BUILD_VARIANT=userdebug
TARGET_BUILD_TYPE=release
TARGET_ARCH=arm64
TARGET_ARCH_VARIANT=armv8-a
TARGET_CPU_VARIANT=generic
TARGET_2ND_ARCH=arm
TARGET_2ND_ARCH_VARIANT=armv7-a-neon
TARGET_2ND_CPU_VARIANT=generic
HOST_ARCH=x86_64
HOST_2ND_ARCH=x86
HOST_OS=linux
HOST_OS_EXTRA=Linux-6.1.0-27-amd64-x86_64-Debian-GNU/Linux-12-(bookworm)
HOST_CROSS_OS=windows
HOST_CROSS_ARCH=x86
HOST_CROSS_2ND_ARCH=x86_64
HOST_BUILD_TYPE=release
BUILD_ID=SP2A.220405.004
OUT_DIR=out
============================================
[ 78% 246/313] including bootable/recovery/Android.mk ...
bootable/recovery/prebuilt/Android.mk:443: warning: vendor_hw:  bootable/recovery/prebuilt/relink.sh out/target/product/fogona/recovery/root/vendor/bin/hw 
[ 95% 300/313] including system/sepolicy/Android.mk ...
system/sepolicy/Android.mk:57: warning: BOARD_PLAT_PUBLIC_SEPOLICY_DIR has been deprecated. Use SYSTEM_EXT_PUBLIC_SEPOLICY_DIRS instead.
system/sepolicy/Android.mk:62: warning: BOARD_PLAT_PRIVATE_SEPOLICY_DIR has been deprecated. Use SYSTEM_EXT_PRIVATE_SEPOLICY_DIRS instead.
ninja: no work to do.

#### build completed successfully (01:43 (mm:ss)) ####
```

jasonmerc, 2024-11-19:
Do I need a prebuilt kernel as well?  If so, how do I add it?  When trying to mka bootimage it threw a prebuilt kernel missing error.  I didn't get this with mka recoveryimage.

d1g0l3b1ck, 2024-11-24:
Yes, grab the prebuilt kernel. You also need to add several lines to your BoardConfig.mk for "mka recoveryimage" to compile:

`TARGET_NO_RECOVERY := false
BOARD_USES_RECOVERY_AS_BOOT := false
BOARD_USES_GENERIC_KERNEL_IMAGE := true
BOARD_EXCLUDE_KERNEL_FROM_RECOVERY_IMAGE := true
`

But I still can't get to the TWRP splash screen. There's a lot more that needs to be done with kernel modules, various dependencies and more.

d1g0l3b1ck, 2024-11-24:
I did manage to make a working recovery (kinda) on that TWRP auto builder site (not gonna look it up at the moment)...but the touchscreen doesn't work, no encryption either. So it's a mess.

jasonmerc, 2024-12-03:
Hey man you did get something at least.  Good stuff.  Do you have trees/sources?  We can port something like Bliss Recovery that doesn't require a touchscreen.

d1g0l3b1ck, 2024-12-04:
https://github.com/d1g0l3b1ck/android_device_motorola_fogona